adamgreen/gcc4mbed #22: Test pass for GCC 4.9-2014-q4-major and mbed SDK rev 93

adamgreen, 2015-02-12:
I have updated the [working branch](https://github.com/adamgreen/gcc4mbed/tree/working) to use mbed SDK rev 93 and the latest build of the GCC_ARM toolchain.

I have tested the platforms that I have access to but not the STM ones.
@svkatielee do you want to test the DISCO_F407VG and NUCLEO_F401RE platforms before I merge into the master branch?  You can skip running the install script to install the latest GCC if you don't want to take the bandwidth hit.  I didn't change anything in the makefiles or code for the new compiler but I did have to change some STM linker scripts paths because of changes in the mbed SDK.  I am therefore more interested in having the mbed SDK updates tested as they continue to refactor the STM tree in the mbed SDK.
@weyoui do you want to test the NUCLEO_F411RE platform before I merge into the master branch?

Thanks,

Adam

weyoui, 2015-02-12:
Adam,

Thanks for asking to test NUCLEO_F411RE board with your new released branch.
I would like to test with  NUCLEO_F411RE  board and let you know the test
result.
I am doing my job in week days so I could test this in this weekend.

Thanks
weyoui

adamgreen, 2015-02-12:
@weyoui Thanks for the quick response and helping out with the testing.  Testing this weekend works well with my schedule.  I don't plan to merge into master until next week.

svkatielee, 2015-02-14:
Hi Adam,
I was under sail the past 3 days. When wew arived here, there seems to be almost no internet.
If I can download anything, I would be glad to test it. This is the status of my repo:

/home/larryl/Electronics/STM32/gcc4mbed
nav2$ git remote -v
origin    https://github.com/svkatielee/gcc4mbed.git (fetch)
origin    https://github.com/svkatielee/gcc4mbed.git (push)
upstream    https://github.com/adamgreen/gcc4mbed.git (fetch)
upstream    https://github.com/adamgreen/gcc4mbed.git (push)

/home/larryl/Electronics/STM32/gcc4mbed
nav2$ git status
On branch working
Your branch is ahead of 'upstream/working' by 17 commits.
  (use "git push" to publish your local commits)

nothing to commit, working directory clean

What is the minimum I could download? What git command do I need?
-Larry  From Larry & Trinda Littlefield
S/V Katie Lee - someplace in the Philippines

adamgreen, 2015-02-14:
@svkatielee Welcome back to port :)  Thanks so much for the offer to test.

It appears that you have co-mingled your own changes into the working branch on your repo.  I suspect that rather than trying to merge in my latest working branch, it will be easier for you to just pull down my working branch into a clean branch on your machine with commands like the following:
```
git fetch git@github.com:adamgreen/gcc4mbed.git working:working_test
git checkout working_test
```

Those above commands should pull my latest working branch into a **working_test** branch on your machine and then switch to that branch.  You can test from that branch and once you want to go back to your own, you can issue the following command:
```
git checkout working
```

weyoui, 2015-02-14:
Adam,

I tested this "working" branch today so I share my result.

*. Tested board : NUCLEO_F411RE board
*. gcc version : latest GCC ARM cross compiler (4.9.3 20141119 (release))

1. HelloWorld -> Test result is Okay, LD2 LED on board is blinking as we expected
2. rtos_basic -> Tested looks Okay.
            This test case is using LED1 and LED2 but these two LEDs are same GPIO port in NUCLEO_F411RE board.
            so, it is hard to know two thread is running without problem.
            My suggestion is how about to use printf statement instead of LEDs.             
3. StdIO -> Tested Okay. I could see the printf, scanf, fprintf functions are working well.
             To check this test, I connected UART RX/TX port on the board. :)
             I think we could add  NUCLEO_F411RE in test case's makefile. 
4. Ticker -> Tested Okay. I could check LD2 LED is blinking with the 5 second period.
            For this test result, I should modify the source code.(main.c).
            Because the LED1 and LED2 is the actually same one in NUCLEO_F411RE board.
           I just block the 34 line "led1 = !led1;" in main.c.

I want to test other cases but I could not. I think if I have the SD card board that is selling in internet. This would help to test additional test cases. I hope to have this test board soon. :) 

Thanks,
weyoui

weyoui, 2015-02-14:
More comment,

adamgreen, 2015-02-15:
@weyoui Thanks for the test pass.  Sounds like it is probably ready to go.  Some feedback I have for your comments:
* It isn't really safe to switch rtos_basic to use printf() since such standard C library calls are not currently thread safe in GCC_ARM/newlib.
* Is there only one LED on these NUCLEO boards?  Can another pin be used instead of LED2 for some of the samples on those boards?
* I meant those git fetch steps specifically for @svkatielee to target his current repo scenario.  It didn't work for you since you don't have SSH setup on your machine to use github.  The steps I have should still work using HTTP, just the fetch URL needs to be switched from git/ssh to http one.

```
git fetch https://github.com/adamgreen/gcc4mbed.git working:working_test
git checkout working_test
```
Thanks again,

Adam

weyoui, 2015-02-15:
Adam,

On NUCLEO_F411RE board, there are three LEDs totally.
But user can use one one LED. the others are used for dedicated usages like
power indicator and com port indicator.
I think other NUCLEO boards are same but I am not sure.

Thanks,
weyoui

svkatielee, 2015-02-15:
Hi Adam,Sorry but it looks like I may not be able to test any for a few more days. The nly network access is thru cell phones here. Two carriers, one works good (3G) one not(Edge). I have the one that doesn't. I took 2 hours for yahoo to display these messages. I tried all day yesterday to download the working branch. Never more than 5 minutes before an error. Even tried getting the .zip gives up after 1 Mb every time, when it needs more than 20Mb.
In a couple days I may be able to get a sim card for the other carrier. Are there lots of differences from 411 and the 401/407? If you are in a hurry, maybe the 411 tests are enough?
- Larry

adamgreen, 2015-02-15:
@svkatielee No worries.  The 411 test coverage should be enough for this release.  If you do update to the latest 93 based version later and hit any issues, then let me know.

Happy Sailing!

closed 2015-02-15 by adamgreen

weyoui, 2015-02-16:
Adam,

I am testing another board, DISCO_F407VG.

HelloWorld is working well.
But StdIO example print out crashed characters.

"5�#%;5=#7��u#5���뿿��M!='5�=������W5�9!%'55��(�忽y�\^^Y�֯��׻U�t��U��T*�Y=#7=ʿm�a�U#-�W5�W忿��W5"

This looks like system clock setting is invalid or UART baud rate is wrong.
I set the UART board rate 9600 8N1 and no flow control.
One thing to note, I found SetSysClock() in  system_stm32f4xx.c has changed
in mbed codes tree.

If you have the some clue about this problem, please let me know.

Thanks,
weyoui