adamgreen/gcc4mbed #29: DS1820 not working

KenjutsuGH, 2015-05-26:
Good day,

Firstly, thank you so much for gcc4mbed! I am using it with my Nucleo F411RE on OS X with eClipse. I prefer offline compilation, since I do not always have internet access.

I am trying to get the  DS1820_HelloWorld program to run (https://developer.mbed.org/users/Sissors/code/DS1820_HelloWorld/). I can get it to compile, and I modified main.cpp as follows:

#define DATA_PIN        A0
#include <mbed.h>
#include "DS1820.h"

Serial pc(USBTX, USBRX);
#define BAUD(x)                 pc.baud(x)
#define GETC(x)                 pc.getc(x)
#define PUTC(x)                 pc.putc(x)
#define PUTS(x)                 pc.puts(x)
#define PRINTF(...)             pc.printf(__VA_ARGS__)
#define READABLE(x)             pc.readable(x)

int main() {
    PUTS("In main\r\n");
    DS1820 probe(DATA_PIN);
    PUTS("Starting while loop...\r\n");
    while(1) {
        probe.convertTemperature(true, DS1820::all_devices); 
        PRINTF("It is %3.1foC\r\n", probe.temperature());
        wait(1);
    }
}

The PRINTF in the while simply displays "It is oC" over and over.

Any ideas?

Thank you

ohagendorf, 2015-05-26:
You can check the linker option for printf/scanf float support. I've already seen several times a space within this option. And with that the option is ignored and the result looks like yours.

Olaf
-- 
Diese Nachricht wurde von meinem Android-Mobiltelefon mit K-9 Mail gesendet.

KenjutsuGH, 2015-05-26:
Thank you @ohagendorf ! That was it. In the makefile, I made the following two changes:

NO_FLOAT_SCANF  := 0
NO_FLOAT_PRINTF := 0

and everything is working as expected :+1:

KenjutsuGH, 2015-05-26:
Just one more question if I may: since I got the DS1820 working, I would like to create a DS1820 library. How would I do that using gcc4mbed? What needs to change in the makefile?

adamgreen, 2015-05-26:
gcc4mbed is currently written to expect library code to just be included in a subdirectory of each project which uses it as happens when you export projects from the mbed site.  The only libraries it supports are the ones from the mbed SDK itself.

closed 2015-05-26 by adamgreen