adamgreen/gcc4mbed #30: NUCLEO_F103RB-device.mk

KenjutsuGH, 2015-06-07:
Hello everyone

Just a repost from the **mbed** forum. Hopefully more **F103RB** users read this post and can test and confirm whether it worked.

Here is my attempt at a **NUCLEO_F103RB-device.mk** file:

```
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

# Vendor/device for which the library should be built.
MBED_DEVICE        := NUCLEO_F103RB
MBED_TARGET        := STM_NUCLEO_F103RB
MBED_CLEAN         := $(MBED_DEVICE)-MBED-clean

# Compiler flags which are specifc to this device.
TARGETS_FOR_DEVICE := TARGET_NUCLEO_F103RB TARGET_M3 TARGET_CORTEX_M TARGET_STM TARGET_STM32F1 TARGET_STM32F103RB
TARGETS_FOR_DEVICE += TARGET_FF_ARDUINO TARGET_FF_MORPHO
GCC_DEFINES := $(patsubst %,-D%,$(TARGETS_FOR_DEVICE))
GCC_DEFINES += -D__CORTEX_M3 -DARM_MATH_CM3 -D__FPU_PRESENT=0

C_FLAGS   := -mcpu=cortex-m3 -mthumb -mfpu=vfp -msoft-float -mthumb-interwork
ASM_FLAGS := -mcpu=cortex-m3 -mthumb -mfpu=vfp -msoft-float
LD_FLAGS  := -mcpu=cortex-m3 -mthumb -mfpu=vfp -msoft-float

# Extra platform specific object files to link into file binary.
DEVICE_OBJECTS :=

# Version of MRI library to use for this device.
DEVICE_MRI_LIB :=

# Linker script to be used.  Indicates what code should be placed where in memory.
LSCRIPT=$(GCC4MBED_DIR)/external/mbed/libraries/mbed/targets/cmsis/TARGET_STM/TARGET_NUCLEO_F103RB/TOOLCHAIN_GCC_ARM/STM32F10X.ld

include $(GCC4MBED_DIR)/build/device-common.mk
```

I do not own such a board, but I decided to give it a go, since I wanted to use the mbed code on a **generic** (eBay) STM32F103 based board. I was able to get it to compile and upload the resulting .hex file to the board using **stm32flash**, and so far it is working

@adamgreen Again, thank for developing **gcc4mbed!** making all of this possible

adamgreen, 2015-06-07:
If you send a pull request with this added file then I can merge it in and prepare it for the next upcoming release of gcc4mbed.

Thanks!

KenjutsuGH, 2015-06-08:
Done :-)

adamgreen, 2015-06-09:
> Done :-)

What has been done?

KenjutsuGH, 2015-06-09:
I'm sorry, I made a total hash of it. I completed the pull request a few moments ago.

adamgreen, 2015-06-09:
Thanks again.  It has been merged into the working branch.

closed 2015-06-09 by adamgreen

LordGuilly, 2017-02-27:
I am currently doing a project using a similar cheap/generic STM32F103C8T6 board (2.5£ on eBay!)
I was planning to use this as a start, but curious what would be the board naming convention, since it's not a NUCLEO platform (GENERIC_F103C8 maybe???)

adamgreen, 2017-02-28:
What kind of changes do you need to make for this generic board? Are they really changes that should go into GCC4MBED or are you really wanting to add a new device to the mbed SDK/HAL proper in https://github.com/ARMmbed/mbed-os ?

LordGuilly, 2017-02-28:
that's an interesting question. The board does not have the debug interface, so it won't fully support the mbed flow by itself - But using an external link interface, I managed to build and download the HelloWorld sample. 
Changes will be required in the linker script (because it's 64Kb flash) and the HAL, as the LED1 is in a different pin. I would think other peripherals may need remapping as well.
I will check the mbed main repo, maybe there is some work in progress there.

LordGuilly, 2017-02-28:
it seems both the mbed SDK and OS support the BLUEPILL_F103C8 target, which is for this board, so once the mbed library moves forward, only the .mk file will be needed.