adamgreen/gcc4mbed #33: No Such File or Directory Install Error

jcady92, 2015-07-06:
I'm installing on Windows 8.1.  I downloaded the compressed installation file and extracted it to C:\gcc4mbed.  Every time I run the win_install.cmd script it downloads the files and then fails when installing.  Here are the logs:

```
Mon 07/06/2015 12:45:18.24  Starting "C:\gcc4mbed\win_install.cmd" 
Mon 07/06/2015 12:45:18.24  Executing external\win32\curl -kL0 https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q1-update/+download/gcc-arm-none-eabi-4_9-2015q1-20150306-win32.zip
Mon 07/06/2015 12:46:23.46  Executing external\win32\md5sum --check C:\gcc4mbed\gcc-arm-none-eabi.md5
gcc-arm-none-eabi-4_9-2015q1-20150306-win32.zip: OK
Mon 07/06/2015 12:46:23.91  Executing rd /s /q C:\gcc4mbed\gcc-arm-none-eabi
Mon 07/06/2015 12:46:23.92  Executing md C:\gcc4mbed\gcc-arm-none-eabi
Mon 07/06/2015 12:46:23.93  Executing cd C:\gcc4mbed\gcc-arm-none-eabi
Mon 07/06/2015 12:46:23.93  Executing ..\external\win32\bsdtar xf C:\gcc4mbed\gcc-arm-none-eabi-4_9-2015q1-20150306-win32.zip
Mon 07/06/2015 12:46:55.80  Executing cd ..
Mon 07/06/2015 12:46:55.84  Executing C:\gcc4mbed\external\win32\make.exe clean
sh: C:\Program: No such file or directory
make: *** [samples.clean] Error 127
Mon 07/06/2015 12:46:55.92  Failure forced early exit
```

Does anyone have any idea what's going on?  Thanks!

adamgreen, 2015-07-06:
I have an idea as to what may be causing the issue.  Let me investigate.

adamgreen, 2015-07-07:
Nope.  That idea didn't pan out.  I shouldn't be explicitly trying to specify a "Program Files.." path from my makefiles.

@jcady92 
The ``sh:`` part of your error message is very unexpected to me.  It is like ``make`` is trying to run a Posix shell even on Windows.  A bit of additional information would help me to track this down:
* How are you running win_install.cmd?  Are you first entering a command shell and then running it or double-clicking on it in Explorer?  If you are using a shell is it cmd.exe or some kind of Posix one like Cygwin?
* From your command shell, can you dump your PATH and paste it in a comment here so that I can see if there is anything obvious grabbing some unexpected attention.
* From your command shell, can you also try dumping the MAKE environment variable to see if it has been set? 
* Even though the install failed, you should have gotten far enough that there is now a ``BuildShell.cmd`` batch file in your gcc4mbed directory.  Can you run that and then issue the following make command manually from the shell that it sets up: ``make VERBOSE=1 clean``  Send me the output of that build attempt if it still fails to complete successfully.
* You can also try going running the ``BuildShell.cmd`` batch file and then go into the samples\HelloWorld directory and issuing a ``make clean`` from there to see if it works even when the build from the root fails.

I have tried to reproduce your problem on my Windows 8.1 install and I can't.  It works as expected.

jcady92, 2015-07-07:
1) I tried clicking on the .cmd in explorer and running it from the cmd.exe shell, both with admin privileges and without.  Same result.

2) Here's my PATH variable:
```
PATH=C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Int
el\Shared Files\cpp\bin\Intel64;C:\Program Files (x86)\Intel\iCLS Client\;C:\Pro
gram Files\Intel\iCLS Client\;C:\windows\system32;C:\windows;C:\windows\System32
\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Intel\Intel(R
) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management En
gine Components\IPT;C:\Program Files (x86)\Intel\Intel(R) Management Engine Comp
onents\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IP
T;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Progra
m Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SD
Ks\TypeScript\1.0\;c:\Program Files (x86)\Microsoft SQL Server\100\Tools\Binn\;c
:\Program Files\Microsoft SQL Server\100\Tools\Binn\;c:\Program Files\Microsoft
SQL Server\100\DTS\Binn\;C:\Program Files\nodejs\;C:\Program Files (x86)\MySQL\M
ySQL Utilities\;C:\Program Files (x86)\MySQL\MySQL Utilities\Doctrine extensions
 for PHP\;C:\Program Files (x86)\GnuWin32\bin;C:\Program Files (x86)\QuickTime\Q
TSystem\;C:\Program Files\TortoiseGit\bin;C:\Program Files (x86)\Nordic Semicond
uctor\nrf51\bin\;C:\Program Files\Calibre2\;C:\Users\jcady_000\AppData\Roaming\n
pm;C:\Program Files (x86)\Git\bin;C:\Program Files (x86)\Git\cmd;C:\Program File
s (x86)\GNU Tools ARM Embedded\4.8 2014q2\bin;C:\Program Files (x86)\GnuWin32\bi
n;C:\Users\jcady_000\AppData\Local\Android\android-studio\sdk\platform-tools;C:\
Users\jcady_000\AppData\Local\Android\android-studio\sdk\tools;C:\apache-ant-1.9
.4\bin;C:\Users\jcady_000\android-sdk\platform-tools;C:\Qt\Qt5.4.0\5.4\mingw491_
32\bin;C:\Users\jcady_000\android-sdk\tools;C:\Users\jcady_000\android-sdk\platf
orm-tools;C:\Python27;C:\Program Files (x86)\CMake\bin;C:\Qt\Qt5.4.1\5.4\mingw49
1_32\bin;C:\Program Files (x86)\CMake\bin
```

3) It doesn't look like I have a MAKE environment variable.

4) Here's the output:
```
sh: C:\Program: No such file or directory
make: *** [samples.clean] Error 127
```

5) I did a `make clean` and it looked to successfully clean the sample.  So I did a `make` and it looks like it successfully built the HelloWorld sample.  Does this mean it installed correctly?

Hope that helps diagnose!

adamgreen, 2015-07-07:
Thanks for the extra information.

>So I did a make and it looks like it successfully built the HelloWorld sample. Does this mean it installed correctly?

Yes, it just means that the makefile in the root isn't working as expected but the project specific ones are.  

I have attempted to make a fix in the working branch to correct this problem.  You could try downloading https://github.com/adamgreen/gcc4mbed/archive/working.zip and see if that installs with no errors.

jcady92, 2015-07-07:
That did the trick!  Installed perfectly.  Thanks Adam!

adamgreen, 2015-07-07:
Ok.  I will get it merged into the master branch as soon as I can.

Thanks for reporting, helping me troubleshoot, and testing.

-Adam

adamgreen, 2015-07-07:
Fixed now merged into master.

Thanks again,

Adam

closed 2015-07-07 by adamgreen