adamgreen/gcc4mbed #34: Including External MBED Library Fails Compilation

jcady92, 2015-07-09:
My target is the NRF51-Dongle (I know it isn't supported, but I added support for it, it's pretty close to the NRF51822).

I am able to compile just fine using gcc4mbed in Eclipse with a simple project, like HelloWorld.  My project is a Bluetooth LE project, so I need to include the BLE_API and NRF51822 libraries that MBED makes.  The issue is that whenever I put those directories into my build directory (without even calling) and attempt to compile, the build fails with a slew of errors.  I've attached the full output of the errors below.

As a test, I exported the BLE_HeartRate test program, which includes those two libraries, for the NRF51-Dongle target to GCC.  I made a makefile project with the MBED generated makefile in Eclipse.  I set the path for arm-none-eabi and it compiled successfully, libraries in all.

It's nice that MBED can export straight to GCC and have everything work out of the box.  But I really don't want to have to copy everything to the online IDE and re-export every time I want to add a file, just so it can generate a proper makefile.  It also doesn't seem to support incremental builds (it builds everything every time, unlike gcc4mbed), which is a bit of a pain.  Are those the main advantages of using gcc4mbed over exporting to gcc from the MBED online compiler?

Do you know how I can include the BLE_API and NRF51822 libraries and have them build properly with gcc4mbed?

```
Description	Resource	Path	Location	Type
'NRF_ERROR_NOT_FOUND' undeclared (first use in this function)	ble_advdata_parser.c	/ble/nRF51822/source/nordic-sdk/components/ble/common	line 55	C/C++ Problem
'NRF_SUCCESS' undeclared (first use in this function)	ble_advdata_parser.c	/ble/nRF51822/source/nordic-sdk/components/ble/common	line 51	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	BLE.h	/ble/BLE_API/ble	line 31	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	BLE.h	/test_proj/BLE_API/ble	line 31	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	BLEInstanceBase.h	/ble/BLE_API/ble	line 31	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	BLEInstanceBase.h	/test_proj/BLE_API/ble	line 31	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	CallChain.h	/HelloWorld/mbed/libraries/mbed/api	line 22	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	CallChain.h	/ble/mbed/libraries/mbed/api	line 22	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	FunctionPointer.h	/HelloWorld/mbed/libraries/mbed/api	line 22	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	FunctionPointer.h	/ble/mbed/libraries/mbed/api	line 22	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	Gap.h	/ble/BLE_API/ble	line 34	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	Gap.h	/test_proj/BLE_API/ble	line 34	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GapAdvertisingData.h	/ble/BLE_API/ble	line 76	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GapAdvertisingData.h	/test_proj/BLE_API/ble	line 76	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GapAdvertisingParams.h	/ble/BLE_API/ble	line 35	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GapAdvertisingParams.h	/test_proj/BLE_API/ble	line 35	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GapEvents.h	/ble/BLE_API/ble	line 30	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GapEvents.h	/test_proj/BLE_API/ble	line 30	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GapScanningParams.h	/ble/BLE_API/ble	line 20	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GapScanningParams.h	/test_proj/BLE_API/ble	line 20	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GattAttribute.h	/ble/BLE_API/ble	line 22	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GattAttribute.h	/test_proj/BLE_API/ble	line 22	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GattCharacteristic.h	/ble/BLE_API/ble	line 26	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GattCharacteristic.h	/test_proj/BLE_API/ble	line 26	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GattClient.h	/ble/BLE_API/ble	line 26	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GattClient.h	/test_proj/BLE_API/ble	line 26	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GattServer.h	/ble/BLE_API/ble	line 27	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GattServer.h	/test_proj/BLE_API/ble	line 27	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GattServerEvents.h	/ble/BLE_API/ble	line 26	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GattServerEvents.h	/test_proj/BLE_API/ble	line 26	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GattService.h	/ble/BLE_API/ble	line 23	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	GattService.h	/test_proj/BLE_API/ble	line 23	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	SecurityManager.h	/ble/BLE_API/ble	line 24	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	SecurityManager.h	/test_proj/BLE_API/ble	line 24	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	ServiceDiscovery.h	/ble/BLE_API/ble	line 27	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	ServiceDiscovery.h	/test_proj/BLE_API/ble	line 27	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	UUID.h	/ble/BLE_API/ble	line 25	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '{' token	UUID.h	/test_proj/BLE_API/ble	line 25	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	CallChainOfFunctionPointersWithContext.h	/ble/BLE_API/ble	line 58	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	CallChainOfFunctionPointersWithContext.h	/test_proj/BLE_API/ble	line 58	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	FunctionPointerWithContext.h	/ble/BLE_API/ble	line 26	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	FunctionPointerWithContext.h	/test_proj/BLE_API/ble	line 26	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	GattCharacteristic.h	/ble/BLE_API/ble	line 457	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	GattCharacteristic.h	/ble/BLE_API/ble	line 466	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	GattCharacteristic.h	/ble/BLE_API/ble	line 475	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	GattCharacteristic.h	/ble/BLE_API/ble	line 485	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	GattCharacteristic.h	/ble/BLE_API/ble	line 495	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	GattCharacteristic.h	/ble/BLE_API/ble	line 505	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	GattCharacteristic.h	/test_proj/BLE_API/ble	line 457	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	GattCharacteristic.h	/test_proj/BLE_API/ble	line 466	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	GattCharacteristic.h	/test_proj/BLE_API/ble	line 475	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	GattCharacteristic.h	/test_proj/BLE_API/ble	line 485	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	GattCharacteristic.h	/test_proj/BLE_API/ble	line 495	C/C++ Problem
expected '=', ',', ';', 'asm' or '__attribute__' before '<' token	GattCharacteristic.h	/test_proj/BLE_API/ble	line 505	C/C++ Problem
[... cut ...]
	$(LD) $(LD_FLAGS) -T$(LINKER_SCRIPT) $(LIBRARY_PATHS) -o $@ $^ $(LIBRARIES) $(LD_SYS_LIBS) $(LIBRARIES) $(LD_SYS_LIBS)

$(PROJECT).bin: $(PROJECT).elf
	$(OBJCOPY) -O binary $< $@

$(PROJECT).hex: $(PROJECT).elf
	@$(OBJCOPY) -O ihex $< $@

$(PROJECT).lst: $(PROJECT).elf
	@$(OBJDUMP) -Sdh $< > $@

lst: $(PROJECT).lst

size: $(PROJECT).elf
	$(SIZE) $(PROJECT).elf

DEPS = $(OBJECTS:.o=.d) $(SYS_OBJECTS:.o=.d)
-include $(DEPS)

merge:
	$(SREC_CAT) $(SOFTDEVICE) -intel $(PROJECT).hex -intel -o combined.hex -intel --line-length=44
```

adamgreen, 2015-07-09:
gcc4mbed samples/ folder has a BLEHearRate sample already.  Can you start with that first?

jcady92, 2015-07-09:
Well, would you look at that.  I completely missed that sample.  That looks like it will be of some help!

It looks like the libs are about a month or so old, though.  So it's still a little worth it to me to figure out how to drop in the newest ones myself.  Do you have any idea what might be going on?

adamgreen, 2015-07-09:
Maybe you need to match the mbed SDK in external with the BLE / nRF51822 libs?  Are you able to build the existing sample for your new NRF51-Dongle?

jcady92, 2015-07-09:
Hmmmm, that is a good idea.  I'll give that a shot.

The sample built no problem.

jcady92, 2015-07-09:
Looks like the problem persists, even with the latest mbed.

adamgreen, 2015-07-09:
If you pull down the latest mbed SDK into the external directory of gcc4mbed, make sure that you do a "make clean-all" before attempting the first build.  That will make sure that any old files get cleaned out.

jcady92, 2015-07-09:
I didn't do that, but the build files for mbed rely in the mbed directory itself, right?  I did delete the entire mbed directory and replace it with the one I pulled down, so that shouldn't be the issue.

adamgreen, 2015-07-09:
Ok.  I guess I don't know what is causing the problem then.  It sounds like you have done what I would have done.

I can look at it when I update gcc4mbed to a newer version of the mbed SDK later in the year and run the accompanying test pass.

jcady92, 2015-07-09:
Sounds good to me, thanks for the support.  For now, the sample fulfills my needs, so I'm satisfied!

adamgreen, 2015-08-09:
I was able to reproduce this problem when I upgraded to the latest mbed sources and I now have a fix in the working branch with commit 61ae5063846cc8412890e051f0a7dd59967ce8ff.  This should be merged into the master branch soon.  Thanks for reporting.

adamgreen, 2015-10-12:
Fix now merged into master.

closed 2015-10-12 by adamgreen