adamgreen/gcc4mbed #35: GDB Cannot View Variables

jcady92, 2015-07-09:
When running the debugger in Eclipse, I'm unable to see the value of any local variables.

I modified the HelloWorld sample to the following:

```
#include "mbed.h"

DigitalOut myled(LED1);

int main() 
{
    int i = 0;
    while(1) 
    {
        myled = i;
        i = 1;
        wait(0.2);
        myled = i;
        i = 0;
        wait(0.2);
    }
}
```

Then I deployed it to my target (NRF51-Dongle).  I started a gdbserver with pyOCD and started debugging using Eclipse.  If I stop at a breakpoint inside the while loop, Eclipse is unable to give me the value for 'i'.  It reports this error on inspection:

```
Failed to execute MI command:
-data-evaluate-expression i
Error message from debugger back end:
Cannot access memory at address 0x20007fdc
```

If I inspect further and view the raw memory, it indeed cannot display any memory locations, giving me `??????` for all of them.  My guess would be that the elf file isn't appropriately getting the symbols.  I'm not sure if it's a problem with gcc4mbed, pyOCD, or Eclipse.  Any idea what the problem might be?

Here's my configuration:

[image]

jcady92, 2015-07-09:
I just made the same program on the mbed online compiler and exported it to GCC.  Booting that up in Eclipse, building, and deploying in the same way to the same target.  I then started the pyOCD GDB server and debugged with the generated elf.  Success!  It was able to correctly inspect the symbols.  So it's looking more like it's a problem with the elf that gcc4mbed is generating.

jcady92, 2015-07-09:
Even more interesting.  I had a couple variables that were "optimized out" from Eclipse with the mbed build.  So I changed the optimization level to O0.  After changing it to O0 the same problem reproduced in my first post with not being able to view any variables with the same errors.  Optimization O1 seems to work fine though (though it optimizes out some variables).  So perhaps not a problem with gcc4mbed, but a problem with gcc optimizations?  Not sure.

adamgreen, 2015-07-10:
Yes, this is an optimization issue and has nothing to do with gcc4mbed.  If you want guaranteed access to variables you unfortunately need to use -O0.  I am not surprised that the ``i`` variable in your above example gets optimized out by a modern compiler.

closed 2015-07-10 by adamgreen

jcady92, 2015-07-10:
I'm not sure that's actually the case, though.  If I was using anything but the -O0 optimization, I could access some variables and it said some were optimized out.  But if I used the -O0 optimization, that was when I couldn't access any variable and it threw this error:

```
Failed to execute MI command:
-data-evaluate-expression i
Error message from debugger back end:
Cannot access memory at address 0x20007fdc
```

That sounds like atypical behavior, correct?

adamgreen, 2015-07-10:
Is that address, 0x20007fdc, valid for your platform?  If so, sounds like a GDB or pyOCD issue.  Sometimes I see errors like that if GDB rounds up a read that takes it just outside of a valid memory region and "set mem inaccessible-by-default off" hasn't been issued to GDB.

jcady92, 2015-07-12:
I'll have to do some further investigating, but it definitely looks like it's not a problem with gcc4mbed itself.  Thanks for the response.

adamgreen, 2015-07-12:
If you ever happen to travel over to the [mbed IRC channel](https://developer.mbed.org/irc/) you can often find me there as **adamgr** in the evenings (Pacific Standard Time).  We might be able to work out the cause of the issue in realtime there.