adamgreen/gcc4mbed #37: Add external sources directory

garyservin, 2015-07-26:
Hi, I'm trying to add an external directory which contains some source files for a project, my project's folder structure is the following

* /project
   * /my_lib
       * my_lib.cpp
        * my_lib.h
    * makefile
    * main.cpp

The makefile has the following

```
PROJECT         := HelloWorld
DEVICES         := LPC1768
GCC4MBED_DIR    := <gcc4mbed_path>
NO_FLOAT_SCANF  := 1
NO_FLOAT_PRINTF := 1

include $(GCC4MBED_DIR)/build/gcc4mbed.mk
```

If my_lib folder is inside each project folder I can compile it fine, but it weights about 5M so it's not practical to have it copied in each directory. I would like to have something like

* /my_projects
    * /my_lib
        * my_lib.cpp
        * my_lib.h
    * /project1
        * makefile
        * main.cpp (which uses my_lib)
    * /project2
        * makefile
        * other_main.cpp (which uses my_lib)

Is there a simple way to do this in the makefiles?

Thanks in advance!

adamgreen, 2015-07-26:
There is currently no support for this in gcc4mbed.  You currently have to do this the way that you originally describe.

I will keep this issue open to consider for the next gcc4mbed release.  It should be possible to do this in a fashion similar to how the mbed SDK libraries are built.

Thanks,

Adam

adamgreen, 2015-09-06:
I have added USER_LIBS support with commit bd4695b2fae2bf96dbf71edb2aeb66de233b0859 in the working branch.  I will close this issue once I have merged into master during the next release.

garyservin, 2015-09-06:
Wow! this is awesome! thanks! =D

adamgreen, 2015-10-12:
Now merged into master.

closed 2015-10-12 by adamgreen

daniel-bub, 2016-02-13:
Hi Adam,

I do have an issue with the USER_LIBS parameter in case a lib contains subfolders with relevant source files.
I just exported the HTTPClient lib from the old mbed website, which uses source-subfolders.
Is there a quick fix  for this?

Thanks,

Daniel