adamgreen/gcc4mbed #39: Linker Warning: cannot find entry symbol Reset_Handler

daniel-bub, 2015-08-27:
Hi,

I'm using gcc4mbed with my u-blox C027 device.

When I compile in "Release" mode everything works fine, but I have issues compiling in "Debugging" mode.
When linking I get the following error:
```c:/gcc4mbed/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: warning: cannot find entry symbol Reset_Handler; defaulting to 00000000```

I compile using a custom device configuration for the u-blox C027, which is based on the ```LPC1768-device.mk``` config:
```
# Copyright 2015 Adam Green (http://mbed.org/users/AdamGreen/)
# Copyright 2015 Daniel Bub
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

# Vendor/device for which the library should be built.
MBED_DEVICE        := UBLOX_C027
MBED_TARGET        := NXP_LPC17XX
MBED_CLEAN         := $(MBED_DEVICE)-MBED-clean

# Compiler flags which are specifc to this device.
TARGETS_FOR_DEVICE := TARGET_LPC1768 TARGET_M3 TARGET_NXP TARGET_LPC176X TARGET_UBLOX_C027 TARGET_CORTEX_M
GCC_DEFINES := $(patsubst %,-D%,$(TARGETS_FOR_DEVICE))
GCC_DEFINES += -D__CORTEX_M3 -DARM_MATH_CM3

C_FLAGS   := -mcpu=cortex-m3 -mthumb -mthumb-interwork
ASM_FLAGS := -mcpu=cortex-m3 -mthumb
LD_FLAGS  := -mcpu=cortex-m3 -mthumb

# Extra platform specific object files to link into file binary.
DEVICE_OBJECTS :=

# Version of MRI library to use for this device.
DEVICE_MRI_LIB := $(GCC4MBED_DIR)/mri/libmri_mbed1768.a

# Linker script to be used.  Indicates what code should be placed where in memory.
LSCRIPT=$(GCC4MBED_DIR)/external/mbed/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X/TOOLCHAIN_GCC_ARM/LPC1768.ld

include $(GCC4MBED_DIR)/build/device-common.mk
```

Is this a know issue and can it stop debug builds from running on the microcontroller?
Right now I have issues getting the debugger to work using this toolchain.

Kind regards,

Daniel

adamgreen, 2015-08-27:
Try changing MBED_TARGET to something like:
```makefile
MBED_TARGET        := NXP_UBLOX_C027
```

This stops it from colliding with the mbed1768 build.  In a release build, I got tons of warnings before I did that.

daniel-bub, 2015-08-28:
Unfortunately this doesn't solve the issue. Instead it causes the same linker warning for "Release"  and "Debug" builds now.

I think the symbol "Reset_Handler" is defined in the file ```gcc4mbed\external\mbed\libraries\mbed\targets\cmsis\TARGET_NXP\TARGET_LPC176X\TOOLCHAIN_GCC_ARM\startup_LPC17xx.s```, but I can't find this file to be used/referenced anywhere during the build.

adamgreen, 2015-08-28:
I can't reproduce the problem on my machine.  It builds fine for Release and Debug builds with no warnings or errors when building the HelloWorld sample.

You might want to try running a "make clean-all" before attempting a Debug build to see if that helps.

My build output:
```
/depots/gcc4mbed/samples/HelloWorld$ make UBLOX_C027 GCC4MBED_TYPE=Debug
Compiling main.cpp
Compiling ../../src/gcc4mbed.c
Compiling ../../external/mbed/libraries/mbed/common/assert.c
Compiling ../../external/mbed/libraries/mbed/common/board.c
Compiling ../../external/mbed/libraries/mbed/common/error.c
Compiling ../../external/mbed/libraries/mbed/common/gpio.c
Compiling ../../external/mbed/libraries/mbed/common/lp_ticker_api.c
Compiling ../../external/mbed/libraries/mbed/common/mbed_interface.c
Compiling ../../external/mbed/libraries/mbed/common/pinmap_common.c
Compiling ../../external/mbed/libraries/mbed/common/rtc_time.c
Compiling ../../external/mbed/libraries/mbed/common/semihost_api.c
Compiling ../../external/mbed/libraries/mbed/common/ticker_api.c
Compiling ../../external/mbed/libraries/mbed/common/us_ticker_api.c
Compiling ../../external/mbed/libraries/mbed/common/wait_api.c
Compiling ../../external/mbed/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X/cmsis_nvic.c
Compiling ../../external/mbed/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X/system_LPC17xx.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/analogin_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/analogout_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/can_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/ethernet_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/gpio_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/gpio_irq_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/i2c_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/pinmap.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/port_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/pwmout_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/rtc_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/serial_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/sleep.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/spi_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/us_ticker.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/TARGET_UBLOX_C027/C027_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/TARGET_UBLOX_C027/mbed_overrides.c
Compiling ../../external/mbed/libraries/mbed/common/BusIn.cpp
Compiling ../../external/mbed/libraries/mbed/common/BusInOut.cpp
Compiling ../../external/mbed/libraries/mbed/common/BusOut.cpp
Compiling ../../external/mbed/libraries/mbed/common/CAN.cpp
Compiling ../../external/mbed/libraries/mbed/common/CallChain.cpp
Compiling ../../external/mbed/libraries/mbed/common/Ethernet.cpp
Compiling ../../external/mbed/libraries/mbed/common/FileBase.cpp
Compiling ../../external/mbed/libraries/mbed/common/FileLike.cpp
Compiling ../../external/mbed/libraries/mbed/common/FilePath.cpp
Compiling ../../external/mbed/libraries/mbed/common/FileSystemLike.cpp
Compiling ../../external/mbed/libraries/mbed/common/I2C.cpp
Compiling ../../external/mbed/libraries/mbed/common/I2CSlave.cpp
Compiling ../../external/mbed/libraries/mbed/common/InterruptIn.cpp
Compiling ../../external/mbed/libraries/mbed/common/InterruptManager.cpp
Compiling ../../external/mbed/libraries/mbed/common/LocalFileSystem.cpp
Compiling ../../external/mbed/libraries/mbed/common/RawSerial.cpp
Compiling ../../external/mbed/libraries/mbed/common/SPI.cpp
Compiling ../../external/mbed/libraries/mbed/common/SPISlave.cpp
Compiling ../../external/mbed/libraries/mbed/common/Serial.cpp
Compiling ../../external/mbed/libraries/mbed/common/SerialBase.cpp
Compiling ../../external/mbed/libraries/mbed/common/Stream.cpp
Compiling ../../external/mbed/libraries/mbed/common/Ticker.cpp
Compiling ../../external/mbed/libraries/mbed/common/Timeout.cpp
Compiling ../../external/mbed/libraries/mbed/common/Timer.cpp
Compiling ../../external/mbed/libraries/mbed/common/TimerEvent.cpp
Compiling ../../external/mbed/libraries/mbed/common/retarget.cpp
Assembling ../../external/mbed/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X/TOOLCHAIN_GCC_ARM/startup_LPC17xx.S
Linking debug library
Linking UBLOX_C027/HelloWorld.elf
Extracting UBLOX_C027/HelloWorld.bin
Extracting UBLOX_C027/HelloWorld.hex
Extracting disassembly to UBLOX_C027/HelloWorld.disasm
   text	   data	    bss	    dec	    hex	filename
  27508	    120	   1340	  28968	   7128	UBLOX_C027/HelloWorld.elf

/depots/gcc4mbed/samples/HelloWorld$ 
```

daniel-bub, 2015-08-28:
Did you use the device file posted above?

adamgreen, 2015-08-28:
Yes

daniel-bub, 2015-08-28:
Which branch are you working on? Master or working?
I ran the same sample build and again the ```startup_LPC17xx.S``` was not mentioned.
I'll do a clean checkout and try it again.

adamgreen, 2015-08-28:
I tried on both master and working but I would recommend that you only use master since the working branch hasn't had a test pass conducted yet.

Yeah I think you are probably right about the assembly language build rules.  I will fix that in the working branch in the near future.

adamgreen, 2015-08-28:
What OS are you using?  Can you provide the complete output from your failed build attempt?

daniel-bub, 2015-08-28:
Here is my setup:
* Windows 7 64-bit
* u-blox C027 - CMSIS-DAP Interface Firmware (23 Apr 2014)
* gcc4mbed on branch master and up to date

And this is the build log:
```
C:\gcc4mbed\samples\HelloWorld>make DEVICES=UBLOX_C027 GCC4MBED_TYPE=Debug
Compiling main.cpp
Compiling ../../src/gcc4mbed.c
Compiling ../../external/mbed/libraries/mbed/common/pinmap_common.c
Compiling ../../external/mbed/libraries/mbed/common/assert.c
Compiling ../../external/mbed/libraries/mbed/common/lp_ticker_api.c
Compiling ../../external/mbed/libraries/mbed/common/rtc_time.c
Compiling ../../external/mbed/libraries/mbed/common/gpio.c
Compiling ../../external/mbed/libraries/mbed/common/mbed_interface.c
Compiling ../../external/mbed/libraries/mbed/common/us_ticker_api.c
Compiling ../../external/mbed/libraries/mbed/common/board.c
Compiling ../../external/mbed/libraries/mbed/common/error.c
Compiling ../../external/mbed/libraries/mbed/common/wait_api.c
Compiling ../../external/mbed/libraries/mbed/common/semihost_api.c
Compiling ../../external/mbed/libraries/mbed/common/ticker_api.c
Compiling ../../external/mbed/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X/system_LPC17xx.c
Compiling ../../external/mbed/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X/cmsis_nvic.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/serial_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/pinmap.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/pwmout_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/i2c_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/us_ticker.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/rtc_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/can_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/port_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/analogin_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/sleep.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/gpio_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/gpio_irq_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/spi_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/ethernet_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/analogout_api.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/TARGET_UBLOX_C027/mbed_overrides.c
Compiling ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/TARGET_UBLOX_C027/C027_api.c
Compiling ../../external/mbed/libraries/mbed/common/Timeout.cpp
Compiling ../../external/mbed/libraries/mbed/common/InterruptManager.cpp
Compiling ../../external/mbed/libraries/mbed/common/LocalFileSystem.cpp
Compiling ../../external/mbed/libraries/mbed/common/FileLike.cpp
Compiling ../../external/mbed/libraries/mbed/common/SerialBase.cpp
Compiling ../../external/mbed/libraries/mbed/common/BusOut.cpp
Compiling ../../external/mbed/libraries/mbed/common/BusIn.cpp
Compiling ../../external/mbed/libraries/mbed/common/RawSerial.cpp
Compiling ../../external/mbed/libraries/mbed/common/InterruptIn.cpp
Compiling ../../external/mbed/libraries/mbed/common/TimerEvent.cpp
Compiling ../../external/mbed/libraries/mbed/common/SPISlave.cpp
Compiling ../../external/mbed/libraries/mbed/common/Ethernet.cpp
Compiling ../../external/mbed/libraries/mbed/common/FileBase.cpp
Compiling ../../external/mbed/libraries/mbed/common/FileSystemLike.cpp
Compiling ../../external/mbed/libraries/mbed/common/CAN.cpp
Compiling ../../external/mbed/libraries/mbed/common/Stream.cpp
Compiling ../../external/mbed/libraries/mbed/common/Timer.cpp
Compiling ../../external/mbed/libraries/mbed/common/FilePath.cpp
Compiling ../../external/mbed/libraries/mbed/common/BusInOut.cpp
Compiling ../../external/mbed/libraries/mbed/common/I2CSlave.cpp
Compiling ../../external/mbed/libraries/mbed/common/retarget.cpp
Compiling ../../external/mbed/libraries/mbed/common/I2C.cpp
Compiling ../../external/mbed/libraries/mbed/common/CallChain.cpp
Compiling ../../external/mbed/libraries/mbed/common/SPI.cpp
Compiling ../../external/mbed/libraries/mbed/common/Ticker.cpp
Compiling ../../external/mbed/libraries/mbed/common/Serial.cpp
Linking debug library
Linking UBLOX_C027/HelloWorld.elf
c:/gcc4mbed/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: warning: cannot find entry symbol Reset_Handler; defaulting to 00000000
[... 16 lines cut ...]
Compiling ../../src/gcc4mbed.c
mkdir UBLOX_C027\ >nul 2>nul & exit 0
arm-none-eabi-gcc -O0 -mcpu=cortex-m3 -mthumb -mthumb-interwork -g3 -ffunction-sections -fdata-sections -fno-exceptions -fno-delete-null-pointer-checks -fomit-frame-pointer -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wno-missing-braces -DTARGET_LPC1768 -DTARGET_M3 -DTARGET_NXP -DTARGET_LPC176X -DTARGET_UBLOX_C027 -DTARGET_CORTEX_M -D__CORTEX_M3 -DARM_MATH_CM3 -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D__MBED__=1 -MMD -MP -std=gnu99  -DMRI_ENABLE=1 -DMRI_INIT_PARAMETERS='"MRI_UART_MBED_USB"' -DMRI_BREAK_ON_INIT=1 -DMRI_SEMIHOST_STDIO=1 -I. -I [... cut ...]
arm-none-eabi-objcopy -O binary UBLOX_C027/HelloWorld.elf UBLOX_C027/HelloWorld.bin
Extracting UBLOX_C027/HelloWorld.hex
arm-none-eabi-objcopy -R .stack -O ihex UBLOX_C027/HelloWorld.elf UBLOX_C027/HelloWorld.hex
Extracting disassembly to UBLOX_C027/HelloWorld.disasm
arm-none-eabi-objdump -d -f -M reg-names-std --demangle UBLOX_C027/HelloWorld.elf >UBLOX_C027/HelloWorld.disasm
arm-none-eabi-size UBLOX_C027/HelloWorld.elf
   text    data     bss     dec     hex filename
  11528     120     572   12220    2fbc UBLOX_C027/HelloWorld.elf
```

Do you need the full verbose output?

daniel-bub, 2015-08-28:
I think I found the cause after adding a lot of ```$(info)``` statements into the Makefiles.

https://github.com/adamgreen/gcc4mbed/blob/master/build/gcc4mbed.mk#L233

Why is the "s" extension only considered to be valid on non Windows systems?
The file ```startup_LPC17xx.s``` has a small "s" on my Windows system after I did a clean checkout of the git repository. Due to the if branch, the file was not considered a valid source and not selected by the ```find_srcs``` function.

adamgreen, 2015-08-28:
Under Windows the file system is case insensitive.  You originally mentioned that the build was successful for Release builds.  If the Release build assembles the file with .s as an extension, it should also assemble it for Debug builds.  Is it assembling the .s file successfully in Release builds but not Debug builds?

Are you using the version of GNU make in external\win32?  Maybe you have a weird version of make in your path which tries to makes things more Posix like on Windows?

daniel-bub, 2015-08-31:
I think you are right. The release builds worked because in the beginning when I started to use gcc4mbed I was indeed using the ```make.exe``` in the ```external\win32``` folder. 
Later I installed the GNU ARM Eclipse Build Tools, which provides another version of make. And this version seems to be case sensitive. (It's bundled with busybox)
Since I already had built a Release version of the Assembly files but no Debug version, the Release builds continued to work while the Debug ones failed.

Whats the issue to have "s" and "S" in the list of source file extensions? Would it cause duplicate rule entries on a case insensitive file system?

adamgreen, 2015-08-31:
Cool, that explains it.

I will add a note to the README to stress using the included version of make.exe on Windows since the gcc4mbed makefiles expect to use this version of GNU make on Windows along with cmd.exe as the shell.

adamgreen, 2015-08-31:
I am re-opening so that I don't forget to update the README and will close once I have completed those changes.

adamgreen, 2015-10-12:
Merged the README update into the master branch.

closed 2015-10-12 by adamgreen