adamgreen/gcc4mbed #40: Problem with adding NRF51_DK device to GGC4MBED

areitsma, 2015-09-01:
Hello, 
I'm trying to build a NRF51_DK program on eclipse. I have followed this guide: https://github.com/adamgreen/gcc4mbed/blob/master/notes/new_devices.creole#adding-new-devices-to-gcc4mbed and used NRF51822 as base device, but i had no succes. 
If i try to build HelloWorld sample for NRF51_DK I get this error:
```
Compiling main.cpp
In file included from ../../external/mbed/libraries/mbed/api/mbed.h:21:0,
                 from main.cpp:4:
../../external/mbed/libraries/mbed/api/platform.h:21:20: fatal error: device.h: No such file or directory
 #include "device.h"
                    ^
compilation terminated.
make: *** [NRF51_DK/main.o] Error 1
```
I'm pretty sure it has something to do with my target devices, but I feel like i did it right.
My NRF5_dk-device.mk
```
# Vendor/device for which the library should be built.
MBED_DEVICE        := NRF51_DK
MBED_TARGET        := NORDIC_NRF51_DK
MBED_CLEAN         := $(MBED_DEVICE)-MBED-clean

# Compiler flags which are specifc to this device.
TARGETS_FOR_DEVICE := TARGET_NRF51_DK TARGET_M0 TARGET_NORDIC TARGET_NRF51_DK TARGET_MCU_NRF51882
TARGETS_FOR_DEVICE += TARGET_MCU_NORDIC_16K TARGET_CORTEX_M
GCC_DEFINES := $(patsubst %,-D%,$(TARGETS_FOR_DEVICE))
GCC_DEFINES += -D__CORTEX_M0 -DARM_MATH_CM0 -DNRF51

C_FLAGS   := -mcpu=cortex-m0 -mthumb -mthumb-interwork
ASM_FLAGS := -mcpu=cortex-m0 -mthumb
LD_FLAGS  := -mcpu=cortex-m0 -mthumb

# Extra platform specific object files to link into file binary.
# For NRF51 parts, we add in the softdevice.
DEVICE_OBJECTS := $(MBED_DEVICE)/s130_nrf51_1.0.0_softdevice.o

# Version of MRI library to use for this device.
DEVICE_MRI_LIB :=

# Linker script to be used.  Indicates what code should be placed where in memory.
LSCRIPT=$(GCC4MBED_DIR)/build/NRF51_DK.ld

include $(GCC4MBED_DIR)/build/device-common.mk

# Rules to build the SoftDevice object file.
$(MBED_DEVICE)/s130_nrf51_1.0.0_softdevice.bin : $(MBED_SRC_ROOT)/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/s130_nrf51822_1_0_0/s130_nrf51_1.0.0_softdevice.hex
	$(Q) $(OBJCOPY) -I ihex -O binary --gap-fill 0xFF $< $@

$(MBED_DEVICE)/s130_nrf51_1.0.0_softdevice.o : $(MBED_DEVICE)/s130_nrf51_1.0.0_softdevice.bin
	$(Q) $(OBJCOPY) -I binary -O elf32-littlearm -B arm --rename-section .data=.SoftDevice $< $@
```
Some more information when running: workspace_tools/build.py -m NRF51_DK -t GCC_ARM -v
```
arm-none-eabi-gcc -std=gnu99 -c 
-Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers 
-fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer 
-mcpu=cortex-m0 -mthumb -O2 
-D__CORTEX_M0 -DTARGET_NRF51_DK -DTARGET_FF_ARDUINO -DTOOLCHAIN_GCC_ARM -DTOOLCHAIN_GCC -DTARGET_MCU_NRF51822 -DTARGET_MCU_NORDIC_32K -DNRF51 -DTARGET_CORTEX_M -DMBED_BUILD_TIMESTAMP=1441024490.28 -DTARGET_NRF51822 -DARM_MATH_CM0 -DTARGET_NORDIC -DTARGET_MCU_NRF51_32K -DTARGET_MCU_NRF51 -DTARGET_M0 -D__MBED__=1 
-I/home/arnoutreitsma/mbed-master/libraries/mbed/common -I/home/arnoutreitsma/mbed-master/build/mbed -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TOOLCHAIN_GCC_ARM -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TARGET_NORDIC -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TARGET_NORDIC/TARGET_MCU_NRF51822 -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TARGET_NORDIC/TARGET_MCU_NRF51822/TARGET_NRF51_DK -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/s130_nrf51822_1_0_0 -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/s110_nrf51822_8_0_0 -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/util -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/crc16 -I/home/arnoutreitsma/mbed-master/build/mbed/TARGET_NRF51_DK/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/scheduler 
-o /home/arnoutreitsma/mbed-master/build/mbed/.temp/TARGET_NRF51_DK/TOOLCHAIN_GCC_ARM/./wait_api.o /home/arnoutreitsma/mbed-master/libraries/mbed/common/wait_api.c
```

I hope you guys can help me.

adamgreen, 2015-09-01:
I changed the TARGETS_FOR_DEVICE a bit and it worked on my machine:
```Makefile
TARGETS_FOR_DEVICE := TARGET_NRF51822 TARGET_M0 TARGET_NORDIC TARGET_NRF51_DK TARGET_MCU_NRF51822
TARGETS_FOR_DEVICE += TARGET_MCU_NORDIC_32K TARGET_CORTEX_M
```

I hope that helps.

areitsma, 2015-09-01:
Thanks for the reply
But after I've tried this it still says that its missing the device.h

adamgreen, 2015-09-01:
Did you copy and paste from my updates or did you just try to edit your pre-existing version to more closely match my updated version?  The reason that I ask is that I fixed a typo or two in your original that you might miss if you don't just copy and paste my fixed version.

areitsma, 2015-09-01:
I did copy those two lines and paste it, so i did not edit it.
This is what my NRF5_dk-device.mk now looks like.
```
# Vendor/device for which the library should be built.
MBED_DEVICE        := NRF51_DK
MBED_TARGET        := NORDIC_NRF51_DK
MBED_CLEAN         := $(MBED_DEVICE)-MBED-clean

# Compiler flags which are specifc to this device.
TARGETS_FOR_DEVICE := TARGET_NRF51822 TARGET_M0 TARGET_NORDIC TARGET_NRF51_DK TARGET_MCU_NRF51822
TARGETS_FOR_DEVICE += TARGET_MCU_NORDIC_32K TARGET_CORTEX_M
GCC_DEFINES := $(patsubst %,-D%,$(TARGETS_FOR_DEVICE))
GCC_DEFINES += -D__CORTEX_M0 -DARM_MATH_CM0 -DNRF51

C_FLAGS   := -mcpu=cortex-m0 -mthumb -mthumb-interwork
ASM_FLAGS := -mcpu=cortex-m0 -mthumb
LD_FLAGS  := -mcpu=cortex-m0 -mthumb

# Extra platform specific object files to link into file binary.
# For NRF51 parts, we add in the softdevice.
DEVICE_OBJECTS := $(MBED_DEVICE)/s130_nrf51_1.0.0_softdevice.o

# Version of MRI library to use for this device.
DEVICE_MRI_LIB :=

# Linker script to be used.  Indicates what code should be placed where in memory.
LSCRIPT=$(GCC4MBED_DIR)/build/NRF51_DK.ld

include $(GCC4MBED_DIR)/build/device-common.mk

# Rules to build the SoftDevice object file.
$(MBED_DEVICE)/s130_nrf51_1.0.0_softdevice.bin : $(MBED_SRC_ROOT)/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/s130_nrf51822_1_0_0/s130_nrf51_1.0.0_softdevice.hex
    $(Q) $(OBJCOPY) -I ihex -O binary --gap-fill 0xFF $< $@

$(MBED_DEVICE)/s130_nrf51_1.0.0_softdevice.o : $(MBED_DEVICE)/s130_nrf51_1.0.0_softdevice.bin
    $(Q) $(OBJCOPY) -I binary -O elf32-littlearm -B arm --rename-section .data=.SoftDevice $< $@
```

adamgreen, 2015-09-01:
Ok.  I don't know what is causing the problems on your machine then.  The key thing I had to update was changing your TARGET_MCU_NRF51882 to the correct TARGET_MCU_NRF51822 and then it found device.h  The other changes I made to just get it closer to what I saw from your Python build run.

areitsma, 2015-09-01:
Could you sent me you're complete NRF51_DK-device.mk, maybe there are some other things that i missed.

adamgreen, 2015-09-01:
```makefile
# Vendor/device for which the library should be built.
MBED_DEVICE        := NRF51_DK
MBED_TARGET        := NORDIC_NRF51_DK
MBED_CLEAN         := $(MBED_DEVICE)-MBED-clean

# Compiler flags which are specifc to this device.
TARGETS_FOR_DEVICE := TARGET_NRF51822 TARGET_M0 TARGET_NORDIC TARGET_NRF51_DK TARGET_MCU_NRF51822
TARGETS_FOR_DEVICE += TARGET_MCU_NORDIC_32K TARGET_CORTEX_M
GCC_DEFINES := $(patsubst %,-D%,$(TARGETS_FOR_DEVICE))
GCC_DEFINES += -D__CORTEX_M0 -DARM_MATH_CM0 -DNRF51

C_FLAGS   := -mcpu=cortex-m0 -mthumb -mthumb-interwork
ASM_FLAGS := -mcpu=cortex-m0 -mthumb
LD_FLAGS  := -mcpu=cortex-m0 -mthumb

# Extra platform specific object files to link into file binary.
# For NRF51 parts, we add in the softdevice.
DEVICE_OBJECTS := $(MBED_DEVICE)/s130_nrf51_1.0.0_softdevice.o

# Version of MRI library to use for this device.
DEVICE_MRI_LIB :=

# Linker script to be used.  Indicates what code should be placed where in memory.
LSCRIPT=$(GCC4MBED_DIR)/build/NRF51822.ld

include $(GCC4MBED_DIR)/build/device-common.mk

# Rules to build the SoftDevice object file.
$(MBED_DEVICE)/s130_nrf51_1.0.0_softdevice.bin : $(MBED_SRC_ROOT)/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/s130_nrf51822_1_0_0/s130_nrf51_1.0.0_softdevice.hex
	$(Q) $(OBJCOPY) -I ihex -O binary --gap-fill 0xFF $< $@

$(MBED_DEVICE)/s130_nrf51_1.0.0_softdevice.o : $(MBED_DEVICE)/s130_nrf51_1.0.0_softdevice.bin
	$(Q) $(OBJCOPY) -I binary -O elf32-littlearm -B arm --rename-section .data=.SoftDevice $< $@
```

**Note:** I didn't have your updated .ld file so I just pointed it back to the one I already had so that I could reproduce your compiler error.  Also you might have to re-indent the make rules towards the end of the file to get them to be tabs instead of spaces since github appears to convert the tabs to spaces.

areitsma, 2015-09-01:
Okay nvm i found the bug, it was a really stupid one. Ty for the help.

One more question, can i configure it that after i build it it automatically get send to the NRF51_DK device?

adamgreen, 2015-09-01:
It depends on how you manually deploy it.  I have to use GDB for my Nordic board but if you are able to deploy successfully by copying the resulting .hex file to the USB mass storage file then you could look at [this documentation](https://github.com/adamgreen/gcc4mbed/blob/master/notes/install.creole#deploy) to see how the makefile can automatically issue the copy if you specify the **NRF51_DK-deploy** rule to GNU make.

closed 2015-09-01 by areitsma