adamgreen/gcc4mbed #42: Debugging won't work 'unrecognized item "timeout" in "qSupported" response'

areitsma, 2015-09-03:
Hi,

I have a NRF51_DK and try to debug with it in eclipse, building and all configurations works and I've updated the firmware of the chip (also tried JLINK firmware).
But I have a problem, when i try to debug a program with gcc4mbed in eclipse, i get this error:

```
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
Ignoring packet error, continuing...
```

I get error window with details:
```
Error in final launch sequence
Failed to execute MI command:
-target-select remote /dev/tty.usbmodem2622
Error message from debugger back end:
Bogus trace status reply from target: timeout
Bogus trace status reply from target: timeout
```

Does someone know a solution?

adamgreen, 2015-09-03:
The fact that you have /dev/tty.usbmodem2622 in the error text indicates that you are probably using the mbed-LPC1768 Eclipse configuration steps which use the MRI debug monitor.  That won't work for your board.  You will have to learn how to use GDB with the debugger hardware on your board and then configure Eclipse to configure GDB the same way.

areitsma, 2015-09-04:
Oke I have found the right configurations for the NRF51_DK and how to debug with it in Eclipse. I now use the JLINK debugger. There just one more problem, when  debug it it i get the error:
```
Error while launching command: gdb --version
```
When i run the command in terminal: gdb --version, It does say its installed
GNU gdb (GDB) 7.10 .....etc
Do you maybe know how to fix this?

adamgreen, 2015-09-08:
Sorry, I don't if there is anyone that reads the issues on this project which can help with your particular setup.  You aren't really hitting a gcc4mbed issue.  Maybe an Eclipse, JLINK, or NRF51_DK specific site might be able to point you in the right direction?

closed 2015-09-17 by adamgreen