adamgreen/gcc4mbed #46: C027 and Eclipse integration

guarndt, 2015-11-12:
Dear Adam and contributors,

I'm trying to add support for the Ublox C027 to gcc4mbed and, along the way, to integrate gcc4mbed snugly with Eclipse.
I have two problems:
1. Code successfully built for LPC1768 does not seem to run on the C027, although they have the same CPU and firmware.
2. I'm not sure if my new Makefile for the C027 is correct - I've read #39, but include paths seem to be an issue, and there's an error message about LPC1768-deploy.

Let me explain in greater detail:

I've read the helpful [cookbook](https://developer.mbed.org/cookbook/eclipse-for-building-and-debugging) and set up my workspace as follows:
* Eclipse 4.4.2 on Windows 7x64, latest CDT.
* Installed the latest [GNU ARM Eclipse](https://gnuarmeclipse.github.io/install) provided by Liviu Ionescu.
* Cloned gcc4mbed git master repo to a folder, ran its Windows installer, and added it as Eclipse project 'gcc4mbed'. This has several advantages: It is under version control from within Eclipse, workspace-relative include paths can be used from new projects, and 'Open declaration' does not only open the file, but select it in the project explorer.
* Chose gcc4mbed's toolchain and build tools globally in Eclipse's C/C++ Builder settings.
* Imported the Makefile-based sample HelloWorld as Eclipse project gcc4mbed_HelloWorld. Deviating from your cookbook, more include paths had to be added to fix indexer issues, as it does not seem to recurse. You can probably best understand my setup looking at the following screenshot:
[image].

As the C027 is the only mbed device available to me, I tried to test the 'Hello World' code built for the LPC1768 with it - it's the same CPU, and I modified the pin according to C027's PinNames.h. I can build it and flash it, but the LED stays off. Here's the verbose build output:

```
13:04:04 **** Build of configuration Debug for project gcc4mbed_HelloWorld ****
make all deploy GCC4MBED_TYPE=Debug 
Compiling main.cpp
Using built-in specs.
COLLECT_GCC=..\..\gcc-arm-none-eabi\bin\arm-none-eabi-g++
Target: arm-none-eabi
Configured with: /home/build/work/GCC-4-9-build/src/gcc/configure --build=i686-linux-gnu --host=i686-w64-mingw32 --target=arm-none-eabi --prefix=/home/build/work/GCC-4-9-build/install-mingw --libexecdir=/home/build/work/GCC-4-9-build/install-mingw/lib --infodir=/home/build/work/GCC-4-9-build/install-mingw/share/doc/gcc-arm-none-eabi/info --mandir=/home/build/work/GCC-4-9-build/install-mingw/share/doc/gcc-arm-none-eabi/man --htmldir=/home/build/work/GCC-4-9-build/install-mingw/share/doc/gcc-arm-none-eabi/html --pdfdir=/home/build/work/GCC-4-9-build/install-mingw/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-headers=yes --with-newlib --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/home/build/work/GCC-4-9-build/install-mingw/arm-none-eabi --with-libiconv-prefix=/home/build/work/GCC-4-9-build/build-mingw/host-libs/usr --with-gmp=/home/build/work/GCC-4-9-build/build-mingw/host-libs/usr --with-mpfr=/home/build/work/GCC-4-9-build/build-mingw/host-libs/usr --with-mpc=/home/build/work/GCC-4-9-build/build-mingw/host-libs/usr --with-isl=/home/build/work/GCC-4-9-build/build-mingw/host-libs/usr --with-cloog=/home/build/work/GCC-4-9-build/build-mingw/host-libs/usr --with-libelf=/home/build/work/GCC-4-9-build/build-mingw/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors' --with-multilib-list=armv6-m,armv7-m,armv7e-m,cortex-m7,armv7-r
Thread model: single
gcc version 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977] (GNU Tools for ARM Embedded Processors) 
COLLECT_GCC_OPTIONS='-O0' '-mcpu=cortex-m3' '-mthumb' '-mthumb-interwork' '-v' '-g3' '-ffunction-sections' '-fdata-sections' '-fno-exceptions' '-fno-delete-null-pointer-checks' '-fomit-frame-pointer' '-Wall' '-Wextra' '-Wno-unused-parameter' '-Wno-missing-field-initializers' '-Wno-missing-braces' '-D' 'TARGET_LPC1768' '-D' 'TARGET_M3' '-D' 'TARGET_NXP' '-D' 'TARGET_LPC176X' '-D' 'TARGET_MBED_LPC1768' '-D' 'TARGET_CORTEX_M' '-D' '__CORTEX_M3' '-D' 'ARM_MATH_CM3' '-D' 'TOOLCHAIN_GCC' '-D' 'TOOLCHAIN_GCC_ARM' '-D' '__MBED__=1' '-MMD' '-MP' '-fno-rtti' '-std=gnu++11' '-D' 'MRI_ENABLE=1' '-D' 'MRI_INIT_PARAMETERS="MRI_UART_MBED_USB"' '-D' 'MRI_BREAK_ON_INIT=1' '-D' 'MRI_SEMIHOST_STDIO=1' '-I' '.' '-I' '.' '-I' './.settings' '-I' '../../mri' '-I' '../../external/mbed/libraries/mbed' '-I' '../../external/mbed/libraries/mbed/api' '-I' '../../external/mbed/libraries/mbed/common' '-I' '../../external/mbed/libraries/mbed/hal' '-I' '../../external/mbed/libraries/mbed/targets' '-I' '../../external/mbed/libraries/mbed/targets/cmsis' '-I' '../../external/mbed/libraries/mbed/targets/cmsis/TARGET_NXP' '-I' '../../external/mbed/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X' '-I' '../../external/mbed/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X/TOOLCHAIN_GCC_ARM' '-I' '../../external/mbed/libraries/mbed/targets/cmsis/TOOLCHAIN_GCC' '-I' '../../external/mbed/libraries/mbed/targets/hal' '-I' '../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP' '-I' '../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X' '-I' '../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/TARGET_MBED_LPC1768' '-c' '-o' 'LPC1768/main.o'
 c:/firedect/git/gcc4mbed/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/4.9.3/cc1plus.exe -quiet -v -I . -I . -I ./.settings -I ../../mri -I ../../external/mbed/libraries/mbed -I ../../external/mbed/libraries/mbed/api -I ../../external/mbed/libraries/mbed/common -I ../../external/mbed/libraries/mbed/hal -I ../../external/mbed/libraries/mbed/targets -I ../../external/mbed/libraries/mbed/targets/cmsis -I ../../external/mbed/libraries/mbed/targets/cmsis/TARGET_NXP -I ../../external/mbed/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X -I ../../external/mbed/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X/TOOLCHAIN_GCC_ARM -I ../../external/mbed/libraries/mbed/targets/cmsis/TOOLCHAIN_GCC -I ../../external/mbed/libraries/mbed/targets/hal -I ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP -I ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X -I ../../external/mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X/TARGET_MBED_LPC1768 -imultilib armv7-m -iprefix c:\firedect\git\gcc4mbed\gcc-arm-none-eabi\bin\../lib/gcc/arm-none-eabi/4.9.3/ -isysroot c:\firedect\git\gcc4mbed\gcc-arm-none-eabi\bin\../arm-none-eabi -MMD LPC1768/main.d -MP -MQ LPC1768/main.o -dD -D__USES_INITFINI__ -D TARGET_LPC1768 -D TARGET_M3 -D TARGET_NXP -D TARGET_LPC176X -D TARGET_MBED_LPC1768 -D TARGET_CORTEX_M -D __CORTEX_M3 -D ARM_MATH_CM3 -D TOOLCHAIN_GCC -D TOOLCHAIN_GCC_ARM -D __MBED__=1 -D MRI_ENABLE=1 -D MRI_INIT_PARAMETERS="MRI_UART_MBED_USB" -D MRI_BREAK_ON_INIT=1 -D MRI_SEMIHOST_STDIO=1 main.cpp -quiet -dumpbase main.cpp -mcpu=cortex-m3 -mthumb -mthumb-interwork -auxbase-strip LPC1768/main.o -g3 -O0 -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wno-missing-braces -std=gnu++11 -version -ffunction-sections -fdata-sections -fno-exceptions -fno-delete-null-pointer-checks -fomit-frame-pointer -fno-rtti -o C:\Users\guarndt\AppData\Local\Temp\ccBzvFeJ.s
GNU C++ (GNU Tools for ARM Embedded Processors) version 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977] (arm-none-eabi)
	compiled by GNU C version 4.7.4, GMP version 4.3.2, MPFR version 2.4.2, MPC version 0.8.1
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring duplicate directory "c:/firedect/git/gcc4mbed/gcc-arm-none-eabi/lib/gcc/../../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/include/c++/4.9.3"
ignoring duplicate directory "c:/firedect/git/gcc4mbed/gcc-arm-none-eabi/lib/gcc/../../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/include/c++/4.9.3/arm-none-eabi/armv7-m"
ignoring duplicate directory "c:/firedect/git/gcc4mbed/gcc-arm-none-eabi/lib/gcc/../../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/include/c++/4.9.3/backward"
ignoring duplicate directory "c:/firedect/git/gcc4mbed/gcc-arm-none-eabi/lib/gcc/../../lib/gcc/arm-none-eabi/4.9.3/include"
[... cut ...]
  27692	    120	   1340	  29152	   71e0	UBLOX_C027/gcc4mbed_C027_HelloWorld.elf
-
make: *** No rule to make target `LPC1768-deploy', needed by `deploy'.  Stop.

14:50:18 Build Finished (took 2s.355ms)
```

Eclipse shows one more error after building, which seems to be an indexer issue only, but is still unexplainable to me:
[image]

Now, thank you so much for reading till here - that's probably already more than I could have expected.
Do you have any idea what I'm doing wrong? I'd be grateful for any suggestion you can make.

adamgreen, 2015-11-12:
I noticed that in your build output the following text `make all deploy GCC4MBED_TYPE=Debug`  I believe that the Eclipse instructions should discuss how to build a Release version as well.  You might want to try that.  The Debug version links in debug monitor support which by default will wait for GDB to connect before it actually lets your program run.  That could explain why you aren't getting a blinking LED since it is still waiting for GDB.  You could also try setting `DEVICE_MRI_LIB := ` in your device specific .mk file to not include the debug monitor in the first place.

guarndt, 2015-11-13:
Adam, thank you for the hint - without MRI, the program runs.

Two (comparably tiny) problems remain:
1. Automatic deployment doesn't work: `make: *** No rule to make target `LPC1768-deploy', needed by `deploy'.  Stop.` I have to copy the bin file manually.
2. Eclipse's indexer looses orientation in C027's `PinNames.h`. Maybe I've done something wrong in 'Paths and Symbols'...

adamgreen, 2015-11-14:
More notes about how to deploy from the gcc4mbed makefile can be found [here](https://github.com/adamgreen/gcc4mbed/blob/master/notes/install.creole#deploy)

I don't use Eclipse enough to be able to help you with your second issue.

guarndt, 2015-11-18:
Ok, I'll try to find something. Thank you.

closed 2015-11-18 by guarndt