adamgreen/gcc4mbed #49: The gcc4mbed usart driver of STM32F429 does not work in CrashCatcher

JaredCJR, 2015-12-08:
I would like to add the support of STM32F429 to the [CrashingApp example in CrashCatcher](https://github.com/adamgreen/CrashCatcher/blob/master/samples/CrashingApp/main.cpp),but the default Serial setting does not work.
It seems like the bug is in the gcc4mbed usart driver of STM32F429.
If I use mri to connect to the device,by typing "backtrace",it will remind me that the stack may corrupted.

NOTE: STM32F429 can run CrashCatcher successfully if the UART is implement by myself.

adamgreen, 2015-12-08:
This sounds like it would be an mbed SDK issue and not a GCC4MBED specific problem.  Does the gcc4mbed StdIO sample work on that STM board?  If it does then try commenting out these 2 lines in the CrashingApp sample: 
https://github.com/adamgreen/CrashCatcher/blob/master/samples/CrashingApp/main.cpp#L19 and 
https://github.com/adamgreen/CrashCatcher/blob/master/samples/CrashingApp/main.cpp#L43.
This will use the default baud rate of 9600 instead of 115200.  If that fixes the problem then maybe you are hitting this issue: https://github.com/mbedmicro/mbed/pull/1432

JaredCJR, 2015-12-08:
The StdIO sample in gcc4mbed works for me.I could get the output in GDB when I use mri to connect to my device.

I can get the "printf output" also in the CrashingAPP of CrashCatcher by means of mri,the "printf output" will print to the GDB host.

I could get the CrashCatcher dump when I use my own implementation of usart.

I expect that I could enter the select menu by mri,and get CrashCatcher dump in other ports.
Due to when the device get into hard fault,it is not able to use semihost printf(priority issue).
Is there anyting wrong with my notion? 

I tried to comment out the two line,but it didn't help to this condition.

adamgreen, 2015-12-08:
When using CrashCatcher, you would not want to have MRI linked into your program as well.  Does the StdIO sample work without MRI when you just connect a terminal program to the serial port?  What happens if you build your version of CrashingApp without MRI and connect to it with a terminal program?

JaredCJR, 2015-12-09:
The below description is tested with "Release version",without mri connection.

StdIO sample works for me without mri ,including input and output!
CrashingApp can output the selecting menu on 115200 or 9600(comment out the two lines) baud rate,but it seems that it is not able to get the user input.
It doesn't do anything after I input number.The program will stop due to the "fgets."

If I directly call the [crashWithFPUDisabled](https://github.com/adamgreen/CrashCatcher/blob/master/samples/CrashingApp/main.cpp#L78) after the [last printf](https://github.com/adamgreen/CrashCatcher/blob/master/samples/CrashingApp/main.cpp#L56),I can get the CrashCatcher dump same as the printf output.

The issue turned to the [fgets](https://github.com/adamgreen/CrashCatcher/blob/master/samples/CrashingApp/main.cpp#L57).

I tried to replace the [first lines](https://github.com/adamgreen/CrashCatcher/blob/master/samples/CrashingApp/main.cpp#L57) and [second line](https://github.com/adamgreen/CrashCatcher/blob/master/samples/CrashingApp/main.cpp#L58) with ["scanf",which works in StdIO sample](https://github.com/adamgreen/gcc4mbed/blob/master/samples/StdIO/main.cpp#L30).And it works!
I can select the options ,and get the dump in the same terminal without mri connection.

adamgreen, 2015-12-09:
Maybe your terminal program is not sending a linefeed character when you press the enter key.  You could try pressing CTRL+J to see what happens.

JaredCJR, 2015-12-09:
After input the option,CTRL+J let it works!
"screen" and "neocon" are in the same condition!

Thanks for your advices.

adamgreen, 2015-12-09:
Excellent!

closed 2015-12-09 by adamgreen