adamgreen/gcc4mbed #50: Compiled BIN fails on Freescale FRDM-K64F

ajdolio, 2016-01-29:
I was working with the "HelloWorld" sample which is identical code to the "mbed_blinky" example on the online compiler, however the only difference is that when I compile the "mbed_blinky" online, the code deploys properly with no issues and when I compile the "HelloWorld" and try to deploy it, the code fails and in the fail.txt I get a "RESERVED BITS" error.

adamgreen, 2016-01-29:
What happens if you export your mbed_blinky project from the online compiler to build with the GCC (ARM Embedded) toolchain?  Does it work in that scenario?

How did you use gcc4mbed to target the K64F?  Currently it doesn't support that platform.  Have you added that support?  If so, can you share the contents of your -device.mk file here?

ajdolio, 2016-01-29:
Adam,

Thanks for your reply. I managed to find another article (albeit in German)
explaining how to get the K64F to function with gcc4mbed and it solved my
issue.

Thanks,
Andrew

closed 2016-01-29 by adamgreen