adamgreen/gcc4mbed #52: Fails to make HelloWorld sample

cyndiac, 2016-03-11:
I was trying to deploy some sample code using the following commands in linux (this portion also failed while running the linux_install script):
`cd gcc4mbed/samples/HelloWorld`
`make clean all deploy`

But I keep getting this error:
`../../gcc-arm-none-eabi/bin/arm-none-eabi-g++: 1: ../../gcc-arm-none-eabi/bin/arm-none-eabi-g++: Syntax error: word unexpected (expecting")")`

The mbed is mounted and the gcc4mbed deploy command is exported; this doesn't seem like a setup issue since it's throwing a syntax error.

adamgreen, 2016-03-11:
It seems like it is having some kind of problem running arm-none-eabi-g++ on your Linux distro.  I guess you could try 2 things that might help narrow down the problem:
* what happens if you try just running g++ from your command line:
```
gcc4mbed/gcc-arm-none-eabi/bin/arm-none-eabi-g++ --version
```
* Try running make with the verbose command option and attach the full output here:
```
cd gcc4mbed/samples/HelloWorld
make VERBOSE=1 clean deploy
```

If you are running a 64-bit version of Linux then you will need to install 32-bit binary support if you haven't done that already. On Ubunutu 12.04 I use ```sudo apt-get install ia32-libs``` but on newer versions I think you might need ```sudo apt-get install lib32z1 lib32ncurses5 lib32bz2-1.0 lib32asound2``` to install the equivalent.

cyndiac, 2016-03-12:
Thanks! It makes now. The specific command that I used was 
`sudo apt-get install -y lib32z1 lib32ncurses5 lib32bz2-1.0`
from http://ssb.stsci.edu/ureka/dev/docs/32bit_iraf.html

Apt-get couldn't find the last package you listed, but it seems works fine without it!

closed 2016-03-12 by cyndiac

adamgreen, 2016-03-12:
Happy to hear that helped. I have always seen it give a file not found error before when the 32-bit libs were missing and not that syntax error. Almost looks like the shell tried to run it as a script or something.

Yeah, that last package looked weird to me as well but it was what I saw listed on one website. I will update the README to include the command that worked for you.

Thanks for the follow-up.

adamgreen, 2016-03-12:
I have updated the "Important Notes" in the working branch to include the apt-get command line which worked for you. Will be pushed to master after the next full test pass.

Thanks again!