adamgreen/gcc4mbed #53: gcc4mbed compiler issue

wmmihaa, 2016-04-22:
This question is not about fixing my code, but rather explain why it works to compile (and works) using mbed online but not locally, and if there are any parameters I can use to make it compile the same way it's done using mbed online.

Using gcc4mbed it fails with:
````
MbedJSONValue.h:356:9: error: expected unqualified-id before '--' token
     int getc() {
         ^
MbedJSONValue.h:356:9: error: expected ')' before '--' token
MbedJSONValue.cpp:244:1: error: expected '}' at end of input
  }
  ^
In file included from MbedJSONValue.cpp:1:0:
MbedJSONValue.h:354:51: error: expected unqualified-id at end of input
     input(const char * first, const char * last) {};
                                                   ^
make[2]: *** [NRF51822/MbedJSONValue.o] Error 1
make[1]: *** [MbedJSONValue] Error 2
````
The code segment looks like this:
````
346: class input {
347: protected:
348:     const char * cur_;
349:     const char * end_;
350:     int last_ch_;
351:     bool ungot_;
352:     int line_;
353: public:
354:     input(const char * first, const char * last) : cur_(first), end_(last), last_ch_(-1), ungot_(false), line_(1) {};
355: 
356: int getc() {
357:     if (ungot_) {
358:         ungot_ = false;
359:         return last_ch_;
360:     }
361:     if (cur_ == end_) {
362:         last_ch_ = -1;
363:         return -1;
364:     }
365:     if (last_ch_ == '\n') {
366:         line_++;
367:     }
368:     last_ch_ = *cur_++ & 0xff;
369:     return last_ch_;
370: }
````
Thanks

adamgreen, 2016-04-22:
I haven't really seen those specific errors before so it is kind of hard for me to suggest a fix without having access to all of your code to try compiling/debugging locally.  It should be noted that GCC and the online compiler are different compilers using different standard C libraries so there can be some differences but usually simple changes can be made to code making it portable between the two.

I tried just pulling in the snippet you gave above into the simple HelloWorld sample and it built successfully for me.  You could try the same code on your end to see if it builds for you as well:
```c++
#include "mbed.h"
#include <stdio.h>

DigitalOut myled(LED1);

class input {
protected:
    const char * cur_;
    const char * end_;
    int last_ch_;
    bool ungot_;
    int line_;
public:
    input(const char * first, const char * last) : cur_(first), end_(last), last_ch_(-1), ungot_(false), line_(1) {};
    int getc() {
        if (ungot_) {
            ungot_ = false;
            return last_ch_;
        }
        if (cur_ == end_) {
            last_ch_ = -1;
            return -1;
        }
        if (last_ch_ == '\n') {
            line_++;
        }
        last_ch_ = *cur_++ & 0xff;
        return last_ch_;
    }
};

int main()
{
    const char* test = "Foo";
    input foo(test, test+3);

    foo.getc();
    while(1)
    {
        myled = 1;
        wait(0.2);
        myled = 0;
        wait(0.2);
    }
}
```

wmmihaa, 2016-04-22:
Thank you @adamgreen would you mind testing with the exported project from mbed (I know I'm pushing it ;) )?
[MbedJSONValue_zip_k64f (1).zip](https://github.com/adamgreen/gcc4mbed/files/231786/MbedJSONValue_zip_k64f.1.zip)

adamgreen, 2016-04-22:
Thanks.  I reproduced with your example. As I suspected, newlib Standard C Library defines **getc()** as a macro in stdio.h to improve performance.

One solution is to ```undef getc``` in your header after the ```#include <stdio.h>``` but before declaring your class.

While I had suspected that it was related to such a macro before it didn't repro with my initial attempt that I previously pasted. It turns out that #including mbed.h was working around the problem because it has a ```#include <cstdio>``` instead of ```#include <stdio.h>``` which prefers to place standard C functions in the std:: namespace instead of the global namespace.

So the two solutions are:
* ```#undef getc``` before declaring your class.
* Replace ```#include <stdio.h>``` with ```#include <cstdio>```

I think I prefer the second solution.

wmmihaa, 2016-04-23:
Fantastic! It worked. Thank you so much!

BTW, I'm having some issues with the EthernetInterface, but just found the same lib in the gcc4mbed repo. Should I just use that one instead?

adamgreen, 2016-04-23:
Good to hear that your issue is now resolved!

There is a TCPSocket_HelloWorld in the gcc4mbed/samples folder that you can look at for ethernet based networking. [Its makefile](https://github.com/adamgreen/gcc4mbed/blob/master/samples/TCPSocket_HelloWorld/makefile) shows how to pull in the ethernet library from the mbed SDK using the MBED_LIBS variable.

closed 2016-04-23 by adamgreen

wmmihaa, 2016-04-24:
That did the trick. Thanks you so much for making this public. Works like a charm!

adamgreen, 2016-04-24:
Excellent!

I should point out that you can also use the [export feature in the mbed online compiler](https://developer.mbed.org/handbook/Exporting-to-Make) to create a makefile that should work with GCC_ARM as well. I made [this forum post](https://developer.mbed.org/forum/mbed/topic/2701/?page=5#comment-25951) a few years ago explaining why GCC4MBED still exists even though that export feature is now available on the mbed site.

Happy Coding!