adamgreen/gcc4mbed #54: Install Eclips for ARM

voicevon, 2016-04-28:
I am using window 10.  I followed instruction  till below:
*Set Name: to Eclipse for ARM.
*Set Location: to http://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/
*Click the Ok button.
* There should now be a **_CDT GNU Cross Development Tools_** item listed on the Install dialog. It may take several seconds to populate this item into the list. Once it appears, check it.

My response is:
 **_GNU ARM c/C++ Cross Development Tools_**

So I changed to Ubuntu, and I got the exactly the same response. Is it possible they (eclipse or arm) made any updating on showing name?  Thanks

adamgreen, 2016-04-28:
@voicevon, I don't personally use Eclipse so I am not familiar with the latest steps that it takes to get it working.  Kevin Gordon documented his steps [here](http://www.rescuerobot.org/drupal/?q=content/best-free-offline-ide-mbedorg-projects) (you are probably interested most in the **Installing Eclipse** part about 1/3 of the way down on that page.)

If you do get it working, it would be awesome if you could update https://github.com/adamgreen/gcc4mbed/blob/master/notes/eclipse.creole to make it more up to date.

voicevon, 2016-05-01:
Hi adamgreen,I did it many times on new installed windows7 and Ubuntu, It showed**_ GNU ARM C/C++ Cross Development Tools_** all the way. Since I am newer to Github,I am not sure how to update your document. May I ask you to do it directly?

adamgreen, 2016-05-04:
What exactly are you asking me to change?

voicevon, 2016-05-04:
In your document 
*Set Name: to Eclipse for ARM.
*Set Location: to http://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/
*Click the Ok button.

There should now be a **_CDT GNU Cross Development Tools_** item listed on the Install dialog. It may take several seconds to populate this item into the list. Once it appears, check it.

The real name showed is **_GNU ARM C/C++ Cross Development Tools_**.

adamgreen, 2016-05-04:
Updated in working and master branches.

closed 2016-05-04 by adamgreen