adamgreen/gcc4mbed #58: Getting the USB and SD Card code functioning on a LPC4300

DouglasPearless, 2016-10-10:
Hi Adam,

I am researching how to get the balance of gcc4mbed to fully compile for the LPC4300, mainly the SD Card and USB code on my Bambino210E.

Then I want to get them working for Smoothie2 and I notice that some of the (other) samples compile for the LPC4330_M4, but not the USB code.

Before I start to better understand what I need to do (makefiles, etc), I assume that the LPC4330_M4 support is not (yet) complete and that while I have read the guides on this GitHub site, I haven't missed something obvious ?

Cheers
Douglas

DouglasPearless, 2016-10-11:
I have spent time analysing why the USB code does not compile and have come to the view that the hardware definitions and other key files have not been defined / ported for the LPC4300, is this correct?

arthurwolf, 2016-10-11:
I do believe that's the case yes. The guys who created the bambino are
supposed to provide a mbed port for their board, they did a partial one,
but did not do major things like usb and ethernet.

DouglasPearless, 2016-10-11:
Ah hah!

I will contact Micromint and see what they say.

In the meantime, I have been looking at what is required to get that going ourselves, and I so far have narrowed it down to several files, the USBHAK and USBDescriptors.

Using the LPC4088 as an example (I know it has different USB speed support), then I think I need to write equivalents for:

gcc4mbed/external/mbed/libraries/USBDevice/USBDevice/USBHAL_LPC40.cpp and its equivalent endpoints file (for example USBEndpoints.h or perhaps a new dedicated one USBEndpoints_LPC40.h

There will probably be other files too.

Thoughts?

Cheers
Douglas

arthurwolf, 2016-10-11:
> Ah hah!
>
> I will contact Micromint and see what they say.

I'm already in contact with them about this since recently, please let me
talk to them for now :) I'll report what comes of it.

> In the meantime, I have been looking at what is required to get that going
> ourselves, and I so far have narrowed it down to several files, the USBHAK
> and USBDescriptors.
>
> Using the LPC4088 as an [...]

I think Adam might have some when he comes around here.

>
> Cheers
> Douglas
> >
> > I do believe that's the case yes. The guys who created the bambino are
> > supposed to provide a mbed port for their board, they did a partial one,
> > but did not do major things like usb and ethernet.
> >
> > On Tue, Oct 11, 2016 at 12:12 PM, Douglas Pearless <
> notifications@github.com
> >
> > > I have spent time analysing why the USB code does not compile and have
> > > come to the view that the hardware definitions and other key files
> have not
> > > been defined / ported for the LPC4300, is this correct?
> > >
> > > <https://github.com/adamgreen/gcc4mbed/issues/58#
> issuecomment-252868850>,
> AAGpFR7IxbEWszdTzSQAMWIxxnijdOFyks5qy2D7gaJpZM4KSKSU>
> > >
> >
> >
> >
> > Courage et bonne humeur.
> https://github.com/adamgreen/gcc4mbed/issues/58#issuecomment-252869475>,
> auth/ADkh-xCxBr8AtspnOxFYv0VXj0hVMRT1ks5qy2FVgaJpZM4KSKSU>.

arthurwolf, 2016-10-11:
@triffid maybe you have some thoughts ?

DouglasPearless, 2016-10-11:
OK, 

I had just raised the incomplete libraries with them prior to your email and I am more than happy to wait for your discussions to complete so I won’t follow up with them.

Cheers
Douglas

triffid, 2016-10-12:
Shouldn't be too hard to whip up HAL definitions from the chip datasheet if the rest of the stack is already there

DouglasPearless, 2016-10-12:
I have been reading the sections in the LPC4300 1400++ page manual to better familiar myself with this particular chip, it is heavy going!

Cheers 

Douglas

Links:
------
[1] https://github.com/adamgreen/gcc4mbed/issues/58#issuecomment-253100254
[2] https://github.com/

adamgreen, 2017-01-23:
This is not a gcc4mbed specific issue. Maybe it should be tracked in the Smoothie2 repository instead?

closed 2017-01-23 by adamgreen