adamgreen/gcc4mbed #59: building pre-compiled library using gcc4mbed

awais-ublox, 2017-01-23:
Hello,

I am trying to build ublox wifi odin pre-compiled library from:

mbed-os-master\targets\TARGET_STM\TARGET_STM32F4\TARGET_UBLOX_EVK_ODIN_W2\sdk\TOOLCHAIN_GCC_ARM\libublox-odin-w2-driver.a

through gcc4mbed. I have used LIBS_SUFFIX macro to include it my project.

There are a lot of linker errors since that library references /features/hal , features/netsocket which cannot be directly build as part of mbed SDK.

Is there any way I could build this library using gcc4mbed?

adamgreen, 2017-01-23:
The current version of gcc4mbed only supports the 119 version of the mbed SDK. This is before the switch to mbed OS 5.0.  If you want to build your code from a makefile, using the export feature of the online compiler would probably be a better option.

awais-ublox, 2017-01-24:
Ok thanks Adam.

awais-ublox, 2017-01-24:
Is there any plan to upgrade gcc4mbed for mbedOS5?

adamgreen, 2017-01-24:
I don't have an ETA yet but I do have issue #57 open to track the fact that I need to do it.

closed 2017-01-24 by adamgreen

adamgreen, 2017-05-21:
PS: I recently pushed up changes to the master branch which adds support for mbed-os-5.4.5/release 142.

Awais174, 2017-05-22:
Thanks Adam!