adamgreen/gcc4mbed #61: how to update mbed libary

mdarweash, 2017-03-03:
how to update gcc4mbed to the latest mbed Library ?

any work that we can help in ? or it's straightforward process that we need to learn ?

adamgreen, 2017-03-03:
> can we do this by downloading the mbed library for mbed.org and injecting it ? or there are modifications that are required to be carried out ?

In the past that was possible but the move to mbed OS 5.0 changed things so much that it is no longer possible to do it that way. I will need to rewrite the makefiles to support the latest versions of mbed. I don't currently have an ETA for that work. I am tracking this work item in issue #57.

mdarweash, 2017-03-05:
so my straightforward question is. i've written my NRF51_32K_S130.ld & NRF51822-s130-device.mk to enable s130 device compiling for NRF51 Dongle .... as i need to use it as peripheral and scanner in the same time ....

is this supported by the current setup on gcc4mbed ? as it compiles well with the s130 but still no scanning is working ..every other functions are working properly

[nrf-s130.zip](https://github.com/adamgreen/gcc4mbed/files/819279/nrf-s130.zip)
best regards

adamgreen, 2017-03-06:
Unfortunately I don't have any experience with using the peripheral & central capable S130 soft device so I don't know if it it can be made to work with the version of the mbed SDK that is in use by gcc4mbed.

My recommendation would be to find a sample on the mbed site that exploits the dual capabilities of the S130 soft device, build/test it with the online compiler, and then export the project to GCC ARM Embedded and use that for producing a GCC created firmware binary.

mdarweash, 2017-03-06:
thanks @adamgreen  for your fast reply... i'll test the whole thing and if it works let see how to publish this for the community

closed 2017-03-09 by adamgreen

adamgreen, 2017-05-21:
PS: I recently pushed up changes to the master branch which adds support for mbed-os-5.4.5/release 142.