adamgreen/gcc4mbed #65: free heap calculation error

DouglasPearless, 2017-12-17:
Hi Adam,

I have come across an unusual situation, which I believe relates to your gcc4mbed work, specifically I am getting weird crashes on Smoothie V1 when I try to “play” a g-code file and I have traced it down to no heap, but it looks very strange.

Repeatable situation:
(1) Boot Smoothie.
(2) Go ‘mem' 

```
Smoothie
ok
mem
Unused Heap: 13420 bytes
Used Heap Size: 12524
Allocated: 268450228, Free: 0
Total Free RAM: 13420 bytes
Free AHB0: 9008, AHB1: 16384
Block size: 76 bytes, Tickinfo size: 168 bytes
ok
```

(2) Go ‘mem v' 

```
mem v
Unused Heap: 13420 bytes
Used Heap Size: 12524
  Chunk: 1  Address: 0x10000910  Size: 32  
  Chunk: 2  Address: 0x10000938  Size: 512  
  Chunk: 3  Address: 0x10000B40  Size: 180  
  Chunk: 4  Address: 0x10000BF8  Size: 12  
  Chunk: 5  Address: 0x10000C10  Size: 4  
  Chunk: 6  Address: 0x10000C18  Size: 40  
  Chunk: 7  Address: 0x10000C48  Size: 20  
  Chunk: 8  Address: 0x10000C68  Size: 16  
  Chunk: 9  Address: 0x10000C80  Size: 88  
  Chunk: 10  Address: 0x10000CE0  Size: 16  
  Chunk: 11  Address: 0x10000CF8  Size: 28  
  Chunk: 12  Address: 0x10000D18  Size: 88  
  Chunk: 13  Address: 0x10000D78  Size: 428  
  Chunk: 14  Address: 0x10000F30  Size: 16  
  Chunk: 15  Address: 0x10000F48  Size: 16  
  Chunk: 16  Address: 0x10000F60  Size: 4  
  Chunk: 17  Address: 0x10000F68  Size: 28  
  Chunk: 18  Address: 0x10000F90  Size: 428  
  Chunk: 19  Address: 0x10001140  Size: 8  
  Chunk: 20  Address: 0x10001150  Size: 168  
  Chunk: 21  Address: 0x10001200  Size: 168  
  Chunk: 22  Address: 0x100012B0  Size: 4  
  Chunk: 23  Address: 0x100012C0  Size: 20  
  Chunk: 24  Address: 0x100012D8  Size: 4  
  Chunk: 25  Address: 0x100012E8  Size: 24  
  Chunk: 26  Address: 0x10001308  Size: 28  
  Chunk: 27  Address: 0x10001328  Size: 24  
  Chunk: 28  Address: 0x10001348  Size: 268447824  
Allocated: 268450228, Free: 0
Total Free RAM: 13420 bytes
Free AHB0: 9008, AHB1: 16384
Start: 15184b MemoryPool at 0x2007c4b0
.Chunk at 0x2007c4b0 (  +0): used, 268 bytes
.Chunk at 0x2007c5bc (+268): used, 140 bytes
.Chunk at 0x2007c648 (+408): used, 284 bytes
.Chunk at 0x2007c764 (+692): used, 88 bytes
.Chunk at 0x2007c7bc (+780): used, 12 bytes
.Chunk at 0x2007c7c8 (+792): used, 20 bytes
.Chunk at 0x2007c7dc (+812): used, 72 bytes
.Chunk at 0x2007c824 (+884): used, 432 bytes
.Chunk at 0x2007c9d4 (+1316): used, 1140 bytes
.Chunk at 0x2007ce48 (+2456): used, 144 bytes
.Chunk at 0x2007ced8 (+2600): used, 516 bytes
.Chunk at 0x2007d0dc (+3116): used, 2436 bytes
.Chunk at 0x2007da60 (+5552): used, 36 bytes
.Chunk at 0x2007da84 (+5588): used, 588 bytes
.Chunk at 0x2007dcd0 (+6176): free, 9008 bytes
End: total 15184b, free: 9008b
Start: 16384b MemoryPool at 0x20080000
.Chunk at 0x20080000 (  +0): free, 16384 bytes
End: total 16384b, free: 16384b
Block size: 76 bytes, Tickinfo size: 168 bytes
ok

```
This looks to me that there *may* be a bug in the calculation of the very last chunk (268447824) ??

Thoughts

Cheers
Douglas

adamgreen, 2017-12-17:
I don't think that gcc4mbed contains heap walking code. Why do you think this is a gcc4mbed issue?

DouglasPearless, 2017-12-17:
Wow that was quick!
My understanding (I could be wrong!) is that the heap walking code in Smoothie V1 walks the heap which I (think) is managed by the underlying gcc4mbed malloc (etc) code that is supplied as part of gcc4mbed?

adamgreen, 2017-12-17:
It is supplied by the newlib Standard C Library (the nano version for Smoothie v1, I think). Can you point me to the heap walking code in Smoothie and I will take a quick look at it? It kind of smells like you have a heap corruption but I am not 100% certain how to interpret the output but a look at the code should help.

DouglasPearless, 2017-12-17:
[https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/utils/simpleshell/SimpleShell.cpp](https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/utils/simpleshell/SimpleShell.cpp) line 98 is the function

DouglasPearless, 2017-12-17:
This may help:

```
isChunkFree	int	1	
chunkSize	uint32_t	40	
chunkNext	uint32_t	268437808	
chunkCurr	uint32_t	268437768	
freeCurr	uint32_t	268440468	
heapEnd	uint32_t	268450292	
usedSize	uint32_t	0	
chunkNumber	uint32_t	1	
freeSize	uint32_t	0	

isChunkFree	int	1	
chunkSize	uint32_t	520	
chunkNext	uint32_t	268438328	
chunkCurr	uint32_t	268437808	
freeCurr	uint32_t	268440468	
heapEnd	uint32_t	268450292	
usedSize	uint32_t	32	
chunkNumber	uint32_t	2	
freeSize	uint32_t	0	

isChunkFree	int	1	
chunkSize	uint32_t	188	
chunkNext	uint32_t	268438516	
chunkCurr	uint32_t	268438328	
freeCurr	uint32_t	268440468	
heapEnd	uint32_t	268450292	
usedSize	uint32_t	544	
chunkNumber	uint32_t	3	
freeSize	uint32_t	0	

isChunkFree	int	1	
chunkSize	uint32_t	20	
chunkNext	uint32_t	268438536	
chunkCurr	uint32_t	268438516	
freeCurr	uint32_t	268440468	
heapEnd	uint32_t	268450292	
usedSize	uint32_t	724	
chunkNumber	uint32_t	4	
freeSize	uint32_t	0	

isChunkFree	int	1	
chunkSize	uint32_t	12	
chunkNext	uint32_t	268438548	
chunkCurr	uint32_t	268438536	
freeCurr	uint32_t	268440468	
heapEnd	uint32_t	268450292	
usedSize	uint32_t	736	
chunkNumber	uint32_t	5	
freeSize	uint32_t	0	

isChunkFree	int	1	
chunkSize	uint32_t	48	
chunkNext	uint32_t	268438596	
chunkCurr	uint32_t	268438548	
freeCurr	uint32_t	268440468	
heapEnd	uint32_t	268450292	
usedSize	uint32_t	740	
chunkNumber	uint32_t	6	
freeSize	uint32_t	0	

isChunkFree	int	1	
chunkSize	uint32_t	28	
chunkNext	uint32_t	268438624	
chunkCurr	uint32_t	268438596	
freeCurr	uint32_t	268440468	
heapEnd	uint32_t	268450292	
usedSize	uint32_t	780	
[... 198 lines cut ...]
heapEnd	uint32_t	268450292	
usedSize	uint32_t	2352	
chunkNumber	uint32_t	26	
freeSize	uint32_t	0	

isChunkFree	int	1	
chunkSize	uint32_t	32	
chunkNext	uint32_t	268440388	
chunkCurr	uint32_t	268440356	
freeCurr	uint32_t	268440468	
heapEnd	uint32_t	268450292	
usedSize	uint32_t	2380	
chunkNumber	uint32_t	27	
freeSize	uint32_t	0	

isChunkFree	int	1	
chunkSize	uint32_t	268447832	
chunkNext	uint32_t	536888220	
chunkCurr	uint32_t	268440388	
freeCurr	uint32_t	268440468	
heapEnd	uint32_t	268450292	
usedSize	uint32_t	2404	
chunkNumber	uint32_t	28	
freeSize	uint32_t	0
```

DouglasPearless, 2017-12-17:
Hmm, the "chunkNext	uint32_t	536888220" for the last chunk looks very suspicious to me.

adamgreen, 2017-12-17:
It looks like your heap is corrupted. A buffer overflow/underflow could cause this. Writing to an already freed pointer could as well. Attempting to access the heap from multiple threads could cause it too but I think Smoothie breaks into the debugger if you try to access the heap from an ISR.

If you look at the first two lines:
```
Used Heap Size: 12524
  Chunk: 1  Address: 0x10000910  Size: 32  
```
This tells you that the current heap should go from 0x10000910 to (0x10000910 + 12524 = 0x100039FC).

```
  Chunk: 28  Address: 0x10001348  Size: 268447824  
```
Chunk 28 appears to have had its header corrupted. It isn't even that close to the end of the allocated heap so there are probably chunks after this which you aren't seeing in the walk.

DouglasPearless, 2017-12-17:
Ah ha; I think you analysis is that the header is spot on; I manually stepped through the code (Eclipse+gdb+Segger JLink Pro) and changed the ChunkSize on chunk 28 to 32 (a random number) and then restarted the debugger and it kept walking through the heap;   The question is how the heck do I figure out what went wrong; though it does appear to always be chunk 28.

adamgreen, 2017-12-17:
If you know how to use GDB then I would recommend putting a memory access watchpoint on that chunk 28 header at startup and watch who modifies it.

adamgreen, 2017-12-17:
For example from a GDB command line (if memory serves correct):
```watch *(int*)0x10000910```

DouglasPearless, 2017-12-17:
Cool, I will try that, need to figure out how to do that from Eclipse :-)

DouglasPearless, 2017-12-17:
Hmm, I get:

```
  Chunk: 26  Address: 0x10001308  Size: 28  
  Chunk: 27  Address: 0x10001328  Size: 24  
  Chunk: 28  Address: 0x10001348  Size: 268447824  
```
And from the watch:
```
*(unsigned long int *)0x10001308	unsigned long	268568	
*(unsigned long int *)0x10001328	unsigned long	274616	
*(unsigned long int *)0x10001348	unsigned long	268447832	
```
and the watchpoint is never called for 0x10001348 :-(

DouglasPearless, 2017-12-17:
Correction, I restarted everything and:

```
Thread #1 57005 (Suspended : Watchpoint : *(unsigned long int *)0x10001348)	
	memcpy() at 0x36fd4	
	mbed::FATFileHandle::FATFileHandle() at FATFileHandle.cpp:34 0x9640	
	mbed::FATFileSystem::open() at FATFileSystem.cpp:98 0x9722	
	_open() at stdio.cpp:157 0x31d76	
	_open_r() at 0x37164	
	_fopen_r() at 0x36744	
	file_exists() at utils.cpp:149 0x8a1a	
	Config::Config() at Config.cpp:45 0x4e4a	
	Kernel::Kernel() at Kernel.cpp:70 0x5752	
	init() at main.cpp:102 0x460c	
	main() at main.cpp:270 0x49b2	
```

DouglasPearless, 2017-12-17:
This call is Smoothie is trying to open "/sd/config.txt" being the config file on the SD card; no idea WHY there is a heap corruption because of this

adamgreen, 2017-12-17:
Yes, you will most likely want to just keep letting it run after each watch fires. The last watch that fires (and its corresponding stack) before you hit your crash is likely to be the most suspect.

adamgreen, 2017-12-17:
Once you know which stop is the most interesting, you can restart and ignore all of the watchpoint stops until you hit that interesting one and then start digging in to do some real debugging.

DouglasPearless, 2017-12-17:
I think I may have found the issue, I am creating a vector of tuples (menu_items) and I use menu_items.clear(); to delete them all, that is the last access immediately prior to the call to the memory allocation for the "Play" command where they HardFault is triggered:

```
Thread #1 57005 (Suspended : Watchpoint : *(unsigned long int *)0x10001348)	
	_M_dispose() at basic_string.h:249 0x23968	
	~basic_string() at basic_string.h:539 0x23968	
	Player::play_command() at Player.cpp:275 0x23968	
	Player::on_console_line_received() at Player.cpp:257 0x24174	
	Kernel::call_event() at Kernel.cpp:319 0x5ea2	
```
and
```
	void
	_M_dispose(const _Alloc& __a)
	{
#if _GLIBCXX_FULLY_DYNAMIC_STRING == 0
	  if (__builtin_expect(this != &_S_empty_rep(), false))
#endif
	    {
	      // Be race-detector-friendly.  For more info see bits/c++config.
	      _GLIBCXX_SYNCHRONIZATION_HAPPENS_BEFORE(&this->_M_refcount);
	      if (__gnu_cxx::__exchange_and_add_dispatch(&this->_M_refcount,
							 -1) <= 0)
		{
		  _GLIBCXX_SYNCHRONIZATION_HAPPENS_AFTER(&this->_M_refcount);
		  _M_destroy(__a);
		}
	    }
	}  // XXX MT
```
line 249 is the
`_M_destroy(__a);`

and __a is at:
```
Name : __a
	Details:@0x10007e68
```

The method I use to create the menu_items:
```
void MainMenuScreen::addMenuItem(const MenuItemType& item)
{
    menu_items.push_back(item);
}

void MainMenuScreen::addMenuItem(bool is_selectable, bool default_cursor_position) //name, selectable, default for cursor
{
    addMenuItem(make_tuple(nullptr, nullptr, nullptr, false, is_selectable, default_cursor_position));
}

void MainMenuScreen::addMenuItem(const char *name, bool is_selectable, bool default_cursor_position) //name, selectable, default for cursor
{
	if (name == nullptr) {
		addMenuItem(make_tuple(nullptr, nullptr, nullptr, false, is_selectable, default_cursor_position));
	}else {
		string n(name);
		if(n.size() > 20) {
			n= n.substr(0, 20);
		}
		addMenuItem(make_tuple(strdup(n.c_str()), nullptr, nullptr, false, is_selectable, default_cursor_position));
	}
}

void MainMenuScreen::addMenuItem(const char *name, std::function<void(void)> func)
{
    string n(name);
    if(n.size() > 20) {
        n= n.substr(0, 20);
    }
    addMenuItem(make_tuple(strdup(n.c_str()), func, nullptr, false, true, false));
}

void MainMenuScreen::addMenuItem(const char *name, const char *gcode)
{
    string n(name);
    if(n.size() > 20) {
        n= n.substr(0, 20);
    }
    addMenuItem(make_tuple(strdup(n.c_str()), nullptr, strdup(gcode), true, true, false));
}
```

but here is how I delete them all:

`	menu_items.clear();`

DouglasPearless, 2017-12-17:
I was my disposal of the vector of tuples that was the issue, entirely my fault.  Replaced with and array in the meantime; will come back to this when I have time.
@adamgreen thanks for your help.

closed 2017-12-17 by DouglasPearless

adamgreen, 2017-12-17:
Good to hear that you figured it out.

DouglasPearless, 2017-12-17:
Sigh, the heap walk now spews out an endless lines:

```
Unused Heap: 12636 bytes
Used Heap Size: 14716
  Chunk: 1  Address: 0x10000910  Size: 32  
  Chunk: 2  Address: 0x10000938  Size: 512  
  Chunk: 3  Address: 0x10000B40  Size: 180  
  Chunk: 4  Address: 0x10000BF8  Size: 12  
  Chunk: 5  Address: 0x10000C10  Size: 4  
  Chunk: 6  Address: 0x10000C18  Size: 40  
  Chunk: 7  Address: 0x10000C48  Size: 20  
  Chunk: 8  Address: 0x10000C68  Size: 16  
  Chunk: 9  Address: 0x10000C80  Size: 88  
  Chunk: 10  Address: 0x10000CE0  Size: 16  
  Chunk: 11  Address: 0x10000CF8  Size: 28  
  Chunk: 12  Address: 0x10000D18  Size: 88  
  Chunk: 13  Address: 0x10000D78  Size: 428  
  Chunk: 14  Address: 0x10000F30  Size: 16  
  Chunk: 15  Address: 0x10000F48  Size: 16  
  Chunk: 16  Address: 0x10000F60  Size: 4  
  Chunk: 17  Address: 0x10000F68  Size: 28  
  Chunk: 18  Address: 0x10000F90  Size: 428  
  Chunk: 19  Address: 0x10001140  Size: 8  
  Chunk: 20  Address: 0x10001150  Size: 168  
  Chunk: 21  Address: 0x10001200  Size: 168  
  Chunk: 22  Address: 0x100012B0  Size: 4  
  Chunk: 23  Address: 0x100012C0  Size: 20  
  Chunk: 24  Address: 0x100012D8  Size: 4  
  Chunk: 25  Address: 0x100012E8  Size: 24  
  Chunk: 26  Address: 0x10001308  Size: 28  
  Chunk: 27  Address: 0x10001328  Size: 24  
  Chunk: 28  Address: 0x10001348  Size: 4294967288  
  Chunk: 29  Address: 0x10001348  Size: 4294967288  
  Chunk: 30  Address: 0x10001348  Size: 4294967288  
  Chunk: 31  Address: 0x10001348  Size: 4294967288  
  Chunk: 32  Address: 0x10001348  Size: 4294967288  
  Chunk: 33  Address: 0x10001348  Size: 4294967288  
  Chunk: 34  Address: 0x10001348  Size: 4294967288  
```

At least I don't get a HardFault, but it is still not good.

adamgreen, 2017-12-17:
Looks like the size in the header for chunk 28 is now corrupted (set to 0). You will probably want to watch modifications to its header at 0x10001340 (the header is 8 bytes before the data itself).

DouglasPearless, 2017-12-17:
OK, that will probably be a tomorrow task :-)

DouglasPearless, 2017-12-21:
I have retraced the fault after rebooting and found it now occurs at 0x1000115C, so that less 8 bytes for the header is 0x100054, so I set a watchpoint as *(unsigned long long int *)0x10001154	to cover 8 bytes (64 bits) and the first write access is during the startup phase where the storage is initialised.

I modified the heap walk to stop when it found a 0 length size:

```
  Chunk: 19  Address: 0x10001140  Size: 24  
  Chunk: 20  Address: 0x1000115C  Size: 0   IS CORRUPTED***
```

Then I find this that the following occurs:

```
Thread #1 57005 (Suspended : Watchpoint : *(unsigned long long int *)0x10001154)	
	memcpy() at 0x371f4	
	std::string::_S_copy_chars() at 0x32f14	
	char* std::string::_S_construct<char const*>() at 0x33658	
	std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string() at 0x33694	
	Config::Config() at Config.cpp:37 0x4e6c	
	Kernel::Kernel() at Kernel.cpp:70 0x5752	
	init() at main.cpp:102 0x460c	
	main() at main.cpp:270 0x49b2	

```
which relates to:

   // Config source for */config files
    FileConfigSource *fcs = NULL;
    if( file_exists("/local/config") )
        fcs = new FileConfigSource("/local/config", "local");
    **else if( file_exists("/local/config.txt") )**
        fcs = new FileConfigSource("/local/config.txt", "local");

and single stepping through ARM instructions I can see it has written from 0x1000114C:
`/local/c`
and it has entered the watch point break because it is about to write the 'o' and onwards to "nfig.txt".

I am wondering if something has gone wrong with the previous chunk as that is 24 bytes long and if that starts at 0x10001140 that would end at 0x10001164 which is over the top of the header at 0x1000114C and its data starting at 0x1000115C?

If so, how are the "chunks" are allocated ?

DouglasPearless, 2017-12-21:
Um , how does the HEAP_TAGS work?

adamgreen, 2017-12-21:
https://github.com/Smoothieware/Smoothieware/blob/edge/src/makefile#L12

adamgreen, 2017-12-21:
> I am wondering if something has gone wrong with the previous chunk as that is 24 bytes long and if that starts at 0x10001140 that would end at 0x10001164 which is over the top of the header at 0x1000114C and its data starting at 0x1000115C?

Are you sure that it is this write that actually corrupts the allocation that you are worried about? Has that allocation actually happened yet or are you seeing a write to an allocation that will be later freed and then part of it used for a future allocation that you are actually interested in? 

Until you see the watchpoint fire for filling in the header as part of malloc(), you don't care about writes that touch those particular header bytes. After the malloc(), you wouldn't expect to see it touched again until a free() occurs. If you see something unexpected before the free() then you have probably caught your heap corruption.

DouglasPearless, 2017-12-21:
Hmm, good point. 

I need to trap the free and malloc’s until the corruption.

Will do that tomorrow; time for sleep.