adamgreen/gcc4mbed #66: mbedUpdate

iamshengluchen, 2018-01-12:
when I make the 'mbedUpdate',there occur an error;
"main.cpp:2377:73: error: ‘struct dirent’ has no member named ‘d_namWhat '"
What should I do to solve this problem?

adamgreen, 2018-01-12:
>"main.cpp:2377:73: error: ‘struct dirent’ has no member named ‘d_namWhat '"

```d_namWhat``` isn't an identifier that I see on [line 2377](https://github.com/adamgreen/gcc4mbed/blob/master/mbedUpdater/main.cpp#L2373).  Was the error actually about ```d_name```? What environment (operating system, compiler, etc) are you performing this build on? I have only tested it on macOS with the Xcode command line tools. I would expect it to work on other Posix systems like Linux but I don't know about Windows.

iamshengluchen, 2018-01-15:
sorry, it is ”d_namLen“, My OS is Ubuntu 16.04 and the version of gcc is 5.4.0.

adamgreen, 2018-01-15:
I see. You could try applying this diff and it might help with that particular problem.

```diff
--- a/mbedUpdater/main.cpp
+++ b/mbedUpdater/main.cpp
@@ -2369,6 +2369,7 @@ static bool recurseMbedDirectories(FILE* pFile, const char* pFullDirName, const
     struct dirent* pEntry = NULL;
     while (NULL != (pEntry = readdir(pDir)))
     {
+        size_t nameLength = strlen(pEntry->d_name);
         switch (pEntry->d_type)
         {
         case DT_DIR:
@@ -2376,7 +2377,7 @@ static bool recurseMbedDirectories(FILE* pFile, const char* pFullDirName, const
                 0 != strcmp(pEntry->d_name, "..") && 
                 0 != strcmp(pEntry->d_name, ".git"))
             {
-                pChildPath = (char*)malloc(fullBaseLength + 1 + pEntry->d_namlen + 1);
+                pChildPath = (char*)malloc(fullBaseLength + 1 + nameLength + 1);
                 if (!pChildPath)
                 {
                     fprintf(stderr, "error: Failed to allocate memory for %s/%s\n", pFullDirName, pEntry->d_name);
@@ -2384,7 +2385,7 @@ static bool recurseMbedDirectories(FILE* pFile, const char* pFullDirName, const
                 }
                 sprintf(pChildPath, "%s/%s", pFullDirName, pEntry->d_name);
 
-                pRelativePath = (char*)malloc(relativeBaseLength + 1 + pEntry->d_namlen + 1);
+                pRelativePath = (char*)malloc(relativeBaseLength + 1 + nameLength + 1);
                 if (!pRelativePath)
                 {
                     fprintf(stderr, "error: Failed to allocate memory for %s/%s\n", pRelativeDirName, pEntry->d_name);
@@ -2405,7 +2406,7 @@ static bool recurseMbedDirectories(FILE* pFile, const char* pFullDirName, const
         case DT_REG:
             if (0 == strcmp(pEntry->d_name, ".mbedignore"))
             {
-                pChildPath = (char*)malloc(fullBaseLength + 1 + pEntry->d_namlen + 1);
+                pChildPath = (char*)malloc(fullBaseLength + 1 + nameLength + 1);
                 if (!pChildPath)
                 {
                     fprintf(stderr, "error: Failed to allocate memory for %s/%s\n", pFullDirName, pEntry->d_name);
```

iamshengluchen, 2018-01-15:
Thank you very much. The code work successfully.

adamgreen, 2018-01-15:
Cool. I will commit that change during my next push, no ETA on that though.

closed 2018-01-15 by adamgreen