adamgreen/gcc4mbed #68: mbed-os update?

wuliuxiansheng, 2018-05-16:
Does the mbed-os version fall behind the official mbed-os? Can we just update mbed-os by replacing it?

adamgreen, 2018-05-16:
GCC4MBED currently uses a version of mbed-os from last year.

The following link does discuss how you can potentially upgraded the version of mbed-os used by gcc4mbed but mbed-os might have changed too much in the last year for that to work but you could try it: https://github.com/adamgreen/gcc4mbed/blob/master/notes/new_devices.creole#adding-new-devices-to-gcc4mbed

You might also want to investigate if you can use [mbed-cli](https://docs.mbed.com/docs/mbed-os-handbook/en/latest/dev_tools/cli/) for your offline gcc based builds instead. I believe that the mbed team still officially supports that method of offline builds.

wuliuxiansheng, 2018-05-16:
Are you going to keep maintaining this project? You think I'd better use mbed-cli rather than gcc4mbed for offline builds? I feel gcc4mbed is a pretty nice tool and it also supports rosserial package which is very helpful.

adamgreen, 2018-05-16:
I don't have a current ETA for the next gcc4mbed update.

Why don't you try this:
* Give mbed-cli a try and see if it can be made to work for your rosserial needs.
* If it doesn't work out for you, come back here and let me know. That should be enough incentive for me to upgrade gcc4mbed for you and other ROS users :)

wuliuxiansheng, 2018-05-17:
I checked mbed-cli and I think it should work because it is actually just move your code from online compiler to your local machine which ended up with a pretty big thing. I feel this is not the workflow I'd like to use. With gcc4mbed, we just need to keep our own files for each project not the whole mbed stuff and ros library. Also gcc4mbed gives us more freedom to configure our projects using a simple Makefile and other tools, like Git. I really hope gcc4mbed can be updated continuously and if possible, I'd like to work with you to maintain this project to make it better.

adamgreen, 2018-05-18:
I have a few projects that I want to finish up over the next couple of weeks. Once they are completed, I will take a look at updating this project.

wuliuxiansheng, 2018-05-21:
I'd like to work with you because this is a very useful tool for the community.

chrisalbertson, 2019-02-23:
What advice would you give to someone starting a new project in 2019 using ROS Serial on mbed?  

A new project would have a long life and I can't tie it to a platform that is not updated.     Seem like I need to stick with ROS Serial Arduino.

adamgreen, 2019-02-23:
@chrisalbertson Does the ROS community have a basic mbed application that gcc4mbed is used to build? I just don't know how gcc4mbed is used in the context of ROS. If you could point me to what you are trying to use gcc4mbed to build, I might be able to help you switch to using a build method supported by the mbed team itself.

wuliuxiansheng, 2019-02-23:
> What advice would you give to someone starting a new project in 2019 using ROS Serial on mbed?
> 
> A new project would have a long life and I can't tie it to a platform that is not updated. Seem like I need to stick with ROS Serial Arduino.

wuliuxiansheng, 2019-02-23:
@chrisalbertson rosserial package doesn’t rely on gcc4mbed. You can use official mbed compiling tool to compile your code.

chrisalbertson, 2019-02-23:
Answering two questions:
1) I think the only ROS conation ti gcc4mbed is with ros_serial on the mbed platform.   ROS of course runs on Linux but ros_serial_mbed takes a little bit of ROS, the part used to pass data and beholds it on mbed and uses gcc4med for that.  

2) but now @wuliuxiansheng says it is possible to use "official mbed compiling tool" rather then gcc4mbed.  But is that mbed CLI or the on-line compiler?

Possibly the problem is the documentation.  It seems to be wrong.   It give two methods to set up ros_serial for mbed and both don't work
(1) the URL in the instruction for the on-line compiler is wrong, the URL does not exist.   But there is a similar URL the works for Kinetic and older but there are no updates after 2016.
(2) the instructions for building on the local machine start with "install gcc4mbed"  

If there is a method that works for the current mbed OS version and the current ROS version please let me know.    I will update the documentation myself after I get this to work.

BTW ros_serial Arduino works fine with version up to date.   as a work around I installed an Arduino bootloader in my STM32F board and I've been publishing data to ROS for about 14 hours non-stop so far.   But I would prefer to go back to mbed

chrisalbertson, 2019-02-23:
The problem may be documentation on the ROS web site is out of date?
http://wiki.ros.org/rosserial_mbed/Tutorials/rosserial_mbed%20Setup

If the above is out of date, I'd be willing to fix it after I learn how to
actually set it up myself.     So how do I use both the current mbed OS and
Meodic?

wuliuxiansheng, 2019-02-23:
@chrisalbertson I think you can use online compiler and generate all ros header files by rosserial package for mbed platform.

adamgreen, 2019-02-28:
Is there anything I can do to help out with this? While I am not that familiar with ROS, I have quite a bit of mbed experience.

chrisalbertson, 2019-03-01:
I think the ROS_Serial people are trying are to use the Mbed CLI
compiler and maybe also the on-line compiler.     But I've not seen
any updates docs.

adamgreen, 2019-03-01:
Ok. Just let me know if there is anything I can do to help.

adamgreen, 2021-06-11:
This project is no longer under active development.

closed 2021-06-11 by adamgreen