adamgreen/gcc4mbed #69: gcc4mbed Ubuntu 14.04 Indigo

100315837, 2018-07-12:
Dear Adam,

I am trying to use the gcc4mbed compiler with ROS Indigo and Ubuntu 14.04 and I am having some issues. What do you recommend me to do?

Thanks in advance.

adamgreen, 2018-07-12:
What type of issues? Can you describe what you have done and what has failed?

100315837, 2018-07-12:
I have followed the Quick Start instructions from README (https://github.com/adamgreen/gcc4mbed/tree/gcc4.8-2014-q1_mbed91) and I have done the following:
1) :~/ROS/catkin_dev/src$ git clone https://github.com/adamgreen/gcc4mbed.git 
2) ~/ROS/catkin_dev/src$ cd gcc4mbed
3) ~/ROS/catkin_dev/src/gcc4mbed(master)$ ./linux_install
Here is where the error appears(at the end):
~/ROS/catkin_dev/src/gcc4mbed ~/ROS/catkin_dev/src/gcc4mbed
Logging install results to /home/user/ROS/catkin_dev/src/gcc4mbed/linux_install.log
Downloading GNU Tools for ARM Embedded Processors...
--2018-07-12 18:27:16--  https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-rm/6_1-2017q1/gcc-arm-none-eabi-6-2017-q1-update-linux.tar.bz2
Resolving armkeil.blob.core.windows.net (armkeil.blob.core.windows.net)... 191.235.193.32
Connecting to armkeil.blob.core.windows.net (armkeil.blob.core.windows.net)|191.235.193.32|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 100247592 (96M) [application/octet-stream]
Saving to: ‘gcc-arm-none-eabi-6-2017-q1-update-linux.tar.bz2’

100%[======================================>] 100.247.592  765KB/s   in 2m 13s 

2018-07-12 18:29:30 (736 KB/s) - ‘gcc-arm-none-eabi-6-2017-q1-update-linux.tar.bz2’ saved [100247592/100247592]

Validating md5 signature of GNU Tools for ARM Embedded Processors...
Extracting GNU Tools for ARM Embedded Processors...
Creating helper scripts...
Performing a clean build of the gcc4mbed samples...
jue jul 12 18:27:16 CEST 2018 Starting ./linux_install
jue jul 12 18:27:16 CEST 2018 Executing wget https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-rm/6_1-2017q1/gcc-arm-none-eabi-6-2017-q1-update-linux.tar.bz2
jue jul 12 18:29:30 CEST 2018 Validating md5 signature of GNU Tools for ARM Embedded Processors
jue jul 12 18:29:30 CEST 2018 Executing tar xf /home/user/ROS/catkin_dev/src/gcc4mbed/gcc-arm-none-eabi-6-2017-q1-update-linux.tar.bz2
jue jul 12 18:29:41 CEST 2018 Executing mv /home/user/ROS/catkin_dev/src/gcc4mbed/gcc-arm-none-eabi-6-2017-q1-update /home/user/ROS/catkin_dev/src/gcc4mbed/gcc-arm-none-eabi
jue jul 12 18:29:41 CEST 2018 Executing make -C samples/ clean-all LPC1768
make: Entering directory `/home/user/ROS/catkin_dev/src/gcc4mbed/samples'
Cleaning HelloWorld/KL25Z
Cleaning HelloWorld/K64F
Cleaning HelloWorld/LPC1768
Cleaning HelloWorld/NRF51_DK
Cleaning ../../external/mbed-os/Debug
Cleaning ../../external/mbed-os/Develop
Cleaning ../../external/mbed-os/Release
Cleaning Blink/LPC1768
Cleaning ../../external/mbed-os/Debug
Cleaning ../../external/mbed-os/Develop
Cleaning ../../external/mbed-os/Release
Cleaning Ticker/KL25Z
Cleaning Ticker/K64F
Cleaning Ticker/LPC1768
Cleaning Ticker/NRF51_DK
Cleaning ../../external/mbed-os/Debug
Cleaning ../../external/mbed-os/Develop
Cleaning ../../external/mbed-os/Release
Cleaning StdIO/KL25Z
Cleaning StdIO/K64F
Cleaning StdIO/LPC1768
Cleaning StdIO/NRF51_DK
Cleaning ../../external/mbed-os/Debug
Cleaning ../../external/mbed-os/Develop
Cleaning ../../external/mbed-os/Release
Cleaning SdPerf/K64F
Cleaning SdPerf/LPC1768
Cleaning ../../external/mbed-os/Debug
Cleaning ../../external/mbed-os/Develop
Cleaning ../../external/mbed-os/Release
Cleaning TCPSocket_HelloWorld/K64F
Cleaning TCPSocket_HelloWorld/LPC1768
Cleaning ../../external/mbed-os/Debug
Cleaning ../../external/mbed-os/Develop
Cleaning ../../external/mbed-os/Release
Cleaning USBMouse/KL25Z
Cleaning USBMouse/K64F
Cleaning USBMouse/LPC1768
Cleaning ../../external/mbed-os/features/unsupported/USBDevice/Debug
Cleaning ../../external/mbed-os/features/unsupported/USBDevice/Develop
Cleaning ../../external/mbed-os/features/unsupported/USBDevice/Release
Cleaning ../../external/mbed-os/Debug
Cleaning ../../external/mbed-os/Develop
Cleaning ../../external/mbed-os/Release
Cleaning BLEHeartRate/NRF51_DK
Cleaning ../../external/mbed-os/Debug
Cleaning ../../external/mbed-os/Develop
Cleaning ../../external/mbed-os/Release
**### **make[1]: Entering directory `/home/user/ROS/catkin_dev/src/gcc4mbed/samples/HelloWorld'
Preprocessing ../../external/mbed-os/targets/TARGET_NXP/TARGET_LPC176X/device/TOOLCHAIN_GCC_ARM/LPC1768.ld
make[1]: Leaving directory `/home/user/ROS/catkin_dev/src/gcc4mbed/samples/HelloWorld'
make: Leaving directory `/home/user/ROS/catkin_dev/src/gcc4mbed/samples'
../../gcc-arm-none-eabi/bin/arm-none-eabi-gcc: 1: ../../gcc-arm-none-eabi/bin/arm-none-eabi-gcc: ELF: not found
../../gcc-arm-none-eabi/bin/arm-none-eabi-gcc: 2: ../../gcc-arm-none-eabi/bin/arm-none-eabi-gcc: Syntax error: "(" unexpected
make[1]: *** [LPC1768/HelloWorld.ld] Error 2
make: *** [HelloWorld.LPC1768] Error 2
jue jul 12 18:29:48 CEST 2018 Failure forced early exit****

Thank you for answering so quickly!

adamgreen, 2018-07-12:
Are you running Ubuntu on a 64-bit Intel machine? The GNU Tools for ARM that I download as part of GCC4MBED only work on that platform.

Can you try going into the HelloWorld sample folder and manually running a verbose build?
```
cd gcc4mbed/samples/HelloWorld
make clean LPC1768 VERBOSE=1
```
Send me the output from that build attempt.

100315837, 2018-07-12:
Thank you Adam. My laboratory closes today and It opens again in September.
Thank you very much for being so attentive. I will send you a message again
in September, I would be very pleased for your help.

Thank you very much again.

Best regards,
Jorge

adamgreen, 2018-07-12:
Ok. Reopen the issue once you get back in the lab. Sorry we couldn't figure it out today.

closed 2018-07-12 by adamgreen

100315837, 2018-07-13:
Okey! I Will reopen the issue in September. Dont worry, we'll fix It! Thank
you very much Adam.