adamgreen/gcc4mbed #71: Helloworld led doesn't blink as expected

electronicLight, 2018-11-23:
I'm using Ubuntu 18.04. The installation instructions have worked and the samples have compiled. I copied the file HelloWorld.bin to my LPC1768 and restarted it. The program seems to work, but the led is supposed to blink every 0.2 seconds and for sure it takes more than 1 second. I played with values and It really seemed to work as expected except the frequency.

I'd like to program mbed offline and gcc4mbed seemed the best option but I'm afraid of errors like this.

Somebody knows how to correct this?

adamgreen, 2018-11-23:
I will look into it. I seem to remember fixing an issue like this before.

In the meantime, you could try:
* [Exporting to GCC from mbed online compiler](https://os.mbed.com/docs/v5.10/tools/exporting.html)
* [Using mbed CLI for offline compilation](https://os.mbed.com/docs/v5.10/tools/developing-mbed-cli.html)

adamgreen, 2018-11-30:
@electronicLight I am closing this bug since I can't reproduce the issue. If you get more information that can help me further investigate the issue, feel free to reopen.

closed 2018-11-30 by adamgreen