adamgreen/gcc4mbed #72: Inclusion of FEATURE_LWIP

mateusmenezes95, 2019-06-17:
https://github.com/adamgreen/gcc4mbed/blob/7d79ef307e65f4f913bed655c887a632352c286c/build/ARCH_MAX-device.mk#L11
Should not FEATURE_LWIP be included here?

When I try to compile the code without FEATURE_LWIP for the ARCH_MAX target, I get an error in not including the [EthernetInterface.h](https://github.com/adamgreen/gcc4mbed/blob/master/external/mbed-os/features/FEATURE_LWIP/lwip-interface/EthernetInterface.h). However, when I include FEATURE_LWIP in the `ARCH_MAX-device.mk` file, the code compiles without errors

Error message:
```
Autobuild::Autotools:jiro_firmware_efuse:build: /home/mateus/jiro_firmware_ws/src/jiro_firmware_efuse/src/main.cpp:7:10: fatal error: EthernetInterface.h: No such file or directory
Autobuild::Autotools:jiro_firmware_efuse:build:  #include <EthernetInterface.h>
```

mateusmenezes95, 2019-06-17:
Sorry, I was wrong!

adamgreen, 2021-06-11:
This project is no longer under active development.

closed 2021-06-11 by adamgreen